Ultimaker/Cura #13270: Import profile from G-code sometimes fails—version related but not that simple

daniels220, 2022-09-12:
### Application Version

5.1.0

### Platform

macOS Big Sur

### Printer

Creality CR-10S

### Reproduction steps

1. MAYBE: Select a profile migrated from an older version of Cura.
2. Slice a model and save G-code
3. Open the Profile manager, click Import, and select the G-code file you just saved

### Actual results

At least sometimes, a "Failed to import profile" error dialog with no additional information.

[image]

### Expected results

A profile created from the G-code file.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Aha, the log states pretty much exactly what is wrong:

```
2022-09-11 22:33:51,357 - DEBUG - [MainThread] cura.Settings.CuraContainerRegistry.importProfile [203]: Attempting to import profile /Users/Daniel/Desktop/ClothespinHookBoth.gcode
2022-09-11 22:33:51,359 - INFO - [MainThread] GCodeProfileReader.GCodeProfileReader.read [53]: Attempting to read a profile from the g-code
2022-09-11 22:33:51,370 - WARNING - [MainThread] UM.VersionUpgradeManager.updateFilesData [433]: Unable to upgrade file of type quality_changes of version 4000018
2022-09-11 22:33:51,371 - WARNING - [MainThread] UM.VersionUpgradeManager.updateFilesData [433]: Unable to upgrade file of type quality_changes of version 4000018
2022-09-11 22:33:51,373 - WARNING - [MainThread] cura.Settings.CuraContainerRegistry.addContainer [74]: Instance container creality_cr10s_clothespinhookboth is outdated. Its setting version is 18 but it should be 20.
2022-09-11 22:33:51,373 - WARNING - [MainThread] UM.Settings.ContainerRegistry.removeContainer [615]: Tried to delete container creality_cr10s_clothespinhookboth, which doesn't exist or isn't loaded.
2022-09-11 22:33:51,375 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [442]: file:///Applications/Ultimaker Cura.app/Contents/MacOS/share/cura/resources/qml/Preferences/ProfilesPage.qml:432: ReferenceError: folder is not defined
```

It looks like Cura is saving the file with the wrong declared version number, then freaking out when it tries to load it. Note that some of my profiles in ~/Library/Application Support/cura/5.1/quality_changes have version 18 (having been migrated from Cura 4 and/or Arachne beta), yet work fine. It may be that this version number from the profile is being written to the G-code file, as some of my G-code files have version 20 while others from within the same time period have version 18, and they likely were saved using different profiles.

I can investigate this more carefully if that would be helpful, but my thinking is that Cura should:
1. Update the version number in saved profiles when it migrates the cura/x.y/ folder in (on Mac) Application Support. Since it already keeps a whole separate directory for each Cura version, everything in that directory should match the directory version, right?
2. In any case, always write its own version when creating a G-code file. Saving and immediately re-importing a G-code file should absolutely never fail I would think.
3. Optionally, attempt to read G-code files with an old version (that may have legitimately been created by an older Cura) the same way it migrates profiles from older versions of Cura. It seems like going Cura 4 save G-code -> Cura 5 import from G-code should work as well (which is to say, not perfectly, but just fine for common settings) as Cura 4 save profile -> install Cura 5 and have it migrate the profile.

MariMakes, 2022-09-23:
Hey @daniels220,

Thanks for your report 👍 

This was the first time I'm learning about this workflow 😮 
Thanks for sharing 👨‍🏫 

I had someone take a closer look and it seems to happen because the gcode came from an Arachne Beta/Alpha.
From what I tested the workflows for other Non-Arachne Cura versions seem to be working fine.

Do you maybe have an example of a non-arachne gcode that doesn't work as expected? 
Then we can take another look at it, if we cannot find another example, we'll probably have to defer it. 

Thank you 😄

daniels220, 2022-09-23:
Okay, so I think you're technically right that the problem is due to the involvement of an Arachne beta. However, I think Cura is, first, doing an _awful_ job of error reporting in a way that qualifies as a bug in its own right—the problem here is as much "Cura won't tell me what's wrong" as "Cura won't do what I want/expect"—and second, making some internally-inconsistent decisions about how to handle encountering settings from a beta version.

Looking at the log file, when I start Cura—to be clear, this is _not_ when I try the import, but bear with me—I see the following lines:

```
2022-09-23 11:41:05,883 - WARNING - [MainThread] UM.VersionUpgradeManager.updateFilesData [433]: Unable to upgrade file of type quality_changes of version 4000018
2022-09-23 11:41:05,886 - WARNING - [MainThread] cura.Settings.CuraContainerRegistry.addContainer [74]: Instance container creality_cr10s_bones_ is outdated. Its setting version is 18 but it should be 20.
```

This is because there is a profile in cura/5.1/quality_changes that dates from when I was using the Arachne beta, and when I installed Cura 5.1, it migrated everything from cura/0.0/ to the new cura/5.1/ settings folder it created.

First off, the fact that this error is silent is a problem—if Cura complained on launch that there are profiles it can't upgrade, I would know what to do. Second, the error doesn't _stop_ Cura from loading the profile, or indeed _from updating it_—because I kept my cura/0.0 Arachne-beta settings dir around, and I notice, comparing the profiles, that the version in the 5.1/ folder has lost a couple settings that were removed or renamed since the Arachne beta—5.1 did in fact re-write the file to contain only those settings it understood. This I think is actually a _good_ thing, however, it should also update the version number! Basically, when encountering beta settings, I think a best-effort attempt should be made to update them—"best-effort" meaning "if the resulting settings aren't what you expected, we won't fix it", but it shouldn't leave things in this weird Frankenstein state where they _seem_ to work but cause problems down the line.

Second, if I select such a profile and save G-code, the G-code file includes the old version number from the profile rather than the version number of the Cura that actually wrote it. This seems even more obviously wrong than editing the profile without updating the version number there. Here's an example of this: [v18 profile.gcode.zip](https://github.com/Ultimaker/Cura/files/9635255/v18.profile.gcode.zip). I promise this file was saved using Cura 5.1, however, the selected profile was one that I created using the Arachne beta, then stashed my cura/5.1/ folder so it would re-migrate from the Arachne settings, ending up with a v18 profile as described above.

Now, regarding the original issue of importing from G-code. Presumably, this internally encounters a similar error to the one above—clearly it encounters _some_ error, or the process would succeed. Yet the error dialog gives no details—again, if it said _anything at all_ about the version number, I might realize the problem and be able to resolve it. That seems like a basic issue that has nothing to do with the beta—the error message should give more information about what is actually wrong, that's pretty basic good user experience. But also, Cura is clearly _capable_ of using and even updating v18/Arachne-beta profiles for use with v20/5.1—see above about how profiles created by Arachne _work_ just fine when migrated to 5.1. So, rather than hard-fail, again I think it would be better if importing from v18 G-code made a best-effort attempt to migrate to v20, with the understanding that some settings may get dropped on the floor if they have been renamed, where a stable-version migration (e.g. 4.13 -> 5.1) would take care to preserve them. This is in keeping with the philosophy of beta software, while not being needlessly unhelpful.