Ultimaker/Cura #13271: Infill overlap not respected

PhilBaz, 2022-09-12:
### Application Version

5.1

### Platform

win 10

### Printer

Voron 2.4

### Reproduction steps

Slicing file with individual models settings and overlapping shape settings. Changing infill overlap anywhere does not appear to have an impact.

 

### Actual results

infill overlap appears to remain at default value

### Expected results

Overlap would conform to setting.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[image]
[infill overlap not respected.zip](https://github.com/Ultimaker/Cura/files/9544482/infill.overlap.not.respected.zip)

MariMakes, 2022-09-28:
Hey @PhilBaz,

Thanks for your report 👍 
I'm a bit overwhelmed by your project file. It's been a while since I've seen a user change so many settings in their print job 😮 !

I took a new print job and just changed the infill overlap.
Both in the per object settings and the custom settings. 
[image]
It seems to work correctly for me. 

I've done a lot of clicking in your .3mf file trying to learn which setting was influencing this behavior but couldn't find it. 

We cannot resolve the issue if we cannot reproduce it. 
Do you have an idea what setting might be influencing the infill overlap not being respected?

PhilBaz, 2022-10-01:
@MariMakes 

Yes, I like to dig into the settings. They make all the difference.

Ill dig into it see if I can figure out what combination is causing the problem. 

PS...while I'm here just want to campaign for separate skin overlap settings for a) bridge and b) top surface skin. It is really needed.

alane, 2022-12-13:
[image]
[PI3MK3M_Body_classic.3mf.txt](https://github.com/Ultimaker/Cura/files/10214031/PI3MK3M_Body_classic.3mf.txt)
[image]
[image]

Gotta say I'm having the same problem.  Doesn't seem like changing the overlap does anything, either in the preview or the actual model.  Here are some examples.  Not sure if I should start a new report or if adding to this one is preferred.  Thanks.

smartavionics, 2022-12-13:
That's skin not infill so the infill overlap isn't going to have any visible effect on it.