Ultimaker/Cura #13274: Failed to slice some files, application Hangs when slicing frequently needing restart

wmerritt72, 2022-09-12:
### Application Version

5.1.0

### Platform

MacOs Montery

### Printer

CR-10 Smart

### Reproduction steps

1. Load select Models and change the Quality settings.
2. Attempt to slice model
3. Every other model will hang or present failure to slice error requiring restart of the application and then slicing will commence again
4. 

### Actual results

Files not slicing or application changes requiring a Force Quit

### Expected results

I expected slicing to continue as it worked previously. Never had the issue prior to this week on my M1Pro MacBook pro. There may be some issues with Apple silicon that need to be resolved.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[CURA HANG FILES MAC.zip](https://github.com/Ultimaker/Cura/files/9548006/CURA.HANG.FILES.MAC.zip)

GregValiant, 2022-09-12:
Thanks for the report.
I'm not familiar with a "*.hang" file.   Do I need to rename them?

wmerritt72, 2022-09-12:
I think they are text files. Try renaming them to text.

MariMakes, 2022-09-23:
Hey @wmerritt72,

Welcome to the Ultimaker Cura Github 🚀 

I'm having some trouble openng the files you shared. 
The files are HANG files:
[image]

I've never seen those before!
They are also not in the list of supported file formats. 
https://support.ultimaker.com/hc/en-us/articles/360012475719-What-file-types-are-supported-in-Ultimaker-Cura-

Can you doublecheck for us if you shared the correct files?
We are looking for .3mf files.
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser. 

Without them we cannot help to troubleshoot. 
Thank you.

closed 2022-10-08 by github-actions[bot]

not-marco, 2022-12-29:
.hang files are Mac OS generate analytics reports. You can open them with any text editor as plain text or an RFT. Or you can use Mac OS Console.app to view them.  

Hang files are generated when the event queue for any given application is not responding in a timely manner. It could mean any task is taking longer than expected to be performed, or because that event has crashed. There is no documentation from Apple on how to interpret these reports. A hang does not necessarily contain the resources needed to identify the source of the problem. It is just an assumption the computer makes that an application is hanging or not responding due to a task taking longer than the allowed tolerance.

Hope this helps a bit.