Ultimaker/Cura #13277: Can't 3D Print

omerjohn, 2022-09-12:
### Application Version

5.1.0

### Platform

Windows 10 and 11

### Printer

alfawise U20

### Reproduction steps

After installing Cura 5.0.0 and Update to 5.1.0 my 3D printer WILL NOT PRINT ANYTHING sliced in cura. I even uninstalled both versions and re-installed version 4.6. Will not print anything sliced in any version of cura now. Will not heat bed or nozzle. Only shows my print meter moving on screen and nothing moves. Printer will print files I have sliced prior to installing version 5.0.0 so I know it's not my 3D Printer.

### Actual results

Nothing

### Expected results

Print Project

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[AU20_LockSpool3_1kg_Inland_zip.zip](https://github.com/Ultimaker/Cura/files/9551792/AU20_LockSpool3_1kg_Inland_zip.zip)

GregValiant, 2022-09-13:
Thanks for the report.
I sliced your file without changing anything.  The gcode looks good.  These are the first few lines just before your StartUp Gcode.  (I've added the comments.)
M140 S70 ;Start heating the bed
M105 ;Report Temperatures
M190 S70 ;Wait for the bed to hit 70
M104 S225 ;Start heating the Hot End
M105 ;Report Temperatures
M109 S225 ;Wait for the Hot End to hit 225
Next is your StartUp Gcode with a purge line and then the print starts.  The gcode would print fine on my Ender 3Pro.

So I don't see any problem in the generated Gcode.  Are you printing via the USB, by SD card, or using Octoprint?

omerjohn, 2022-09-14:
I am printinting via an sd card.

GregValiant, 2022-09-14:
Format the SD card to FAT32.  A quick format is fine.  Windows can't tell when sectors of the memory card become corrupted and it will try to write data to those sectors even though they are broken.  Formatting will fix or disable those sectors so Windows will notice and not write there.  I have found that I need to re-format every few months.
If you have additional files on the card that you want to keep you will have to copy them to your hard drive before formatting.

omerjohn, 2022-09-17:
Much THANKS. Formatting the card solved the problem. I never would have
thought of that since it started not working at exactly the same time I
installed the updated software.

Thanks,     Omer

GregValiant, 2022-09-17:
Murphy.  What can go wrong will go wrong and it will happen at the worst possible time.
I'll go ahead and close this as it wasn't a Cura issue.

closed 2022-09-17 by GregValiant