Ultimaker/Cura #13278: Error Slicing with Tree Supports

scott-engemann, 2022-09-13:
### Application Version

5.1.0

### Platform

Windows 10 (64 bit)

### Printer

Creality Ender 3 v2

### Reproduction steps

1. Open STL to print (I was printing the angled STL from this file https://www.thingiverse.com/thing:4831341)
2. Select "Customize" on slicing options
3. Search for "Support"
4. Enable Support
5. Change "Support Structure" from "Normal" to "Tree"
6. Click "Slice"

### Actual results

After a few seconds and while Cura attempts to slice the file, an error occurs.

### Expected results

Cura should successfully finish slicing the STL

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9553922/cura.log)

GregValiant, 2022-09-13:
Thanks for the report.
We need a project file (File | Save Project) to troubleshoot this.
I downloaded the model and didn't have any trouble slicing it.
[image]

scott-engemann, 2022-09-17:
Very odd, because now I'm unable to replicate the error.