Ultimaker/Cura #13282: Brim Outside Build Volume with Tree Support

UxuginPython, 2022-09-13:
### Application Version

5.1.0

### Platform

Fedora 36, Integrated Radeon graphics

### Printer

Creality CR-10

### Reproduction steps

1. Imported the attached model
2. Rescaled it to 14 inches tall
3. Enabled Tree Support, Lightning Infill, and Brim
4. Sliced
5. Went into "Preview"
6. Noticed the issue
[puppy.zip](https://github.com/Ultimaker/Cura/files/9561215/puppy.zip)

### Actual results

Cura generates a brim outside the build volume when using Tree Support

### Expected results

Cura should alert the user that the model cannot be printed at that size without disabling Brim.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

The issue is clearly visible in both Right View and 3D View. Here are screenshots:
[image]
[image]
Here are the project and log:
[CCR10_puppy.3mf.zip](https://github.com/Ultimaker/Cura/files/9561240/CCR10_puppy.3mf.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/9561268/cura.log)
And here's Neofetch in case you need any more system information:
[image]

GregValiant, 2022-09-14:
Thanks for the report.
This behavior has been noted before but I'm unable to find the other bug report.  I am able to reproduce this in Cura 5.1.
I'm going to mark this as a duplicate and someone from the Cura team will take a look.

UxuginPython, 2022-09-14:
@GregValiant OK, Thank you!