Ultimaker/Cura #13284: Halloween mask fail

fgomez37459, 2022-09-14:
### Application Version

5.1.0

### Platform

windows 11

### Printer

snapmaker A350T

### Reproduction steps

1. Sculpted model on my ipad using nomad sculpt.
2. exported to an stl file
3. saved from ipad to usb c type flash drive
4. plugged in flash drive to computer
5. opened cura
6. adjusted layer height to 0.28 mm
7. infill density to 15%
8. print speed to 120 mm/s
9. applyed generate supports with 35 degree overhang 
10. clicked slice and then the program told me there was an unexpected issue

### Actual results

Nothing has happened cannot slice project.

### Expected results

should slice the project

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Mask_2.zip](https://github.com/Ultimaker/Cura/files/9561588/Mask_2.zip)

fgomez37459, 2022-09-14:
Please let me know what i can do to fix this issue.

GregValiant, 2022-09-14:
Thanks for the report.
The model has some errors that I repaired using https://formware.co/OnlineStlRepair.  Here is their report:
--> 6 Naked edges (?)
--> 0 Planar holes (?)
--> 0 Non-planar holes (?)
--> 111 Non-manifold edges (?)
--> 146 Inverted faces (?)
--> 0 Degenerate faces (?)
--> 0 Duplicate faces (?)
--> 0 Disjoint shells (?)
-> Repairing: 100.00%
----- Repair completed in 80953ms ------
-> Vertex count changed from 40849 to 61651 (+20802)
-> Triangle count changed from 81920 to 139824 (+57904)

I couldn't get Cura to slice the repaired model either.  There is a workaround you can try.  Move the model away from the mid-point of the build plate.  Rotate the model 45° or 90°.  The movement and rotation often negate the bug.  In this case though, I couldn't get that to work.
I'll mark this as a duplicate.  This problem is being worked on and should be fixed in 5.2.