Ultimaker/Cura #13289: Failed to slice model with tree supports enabled

bryanduxbury, 2022-09-14:
### Application Version

5.1

### Platform

OSX

### Printer

Voron 2.4 350

### Reproduction steps

Tried to slice a relatively large model with tree supports enabled. After a long slice wait, it failed with an error. cura.log has "Reversing even wall line causes it to be printed CCW instead of CW!" as the last message.

### Actual results

Slicing failed

### Expected results

Slicing succeeded

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9565876/cura.log)
[back.3mf.zip](https://github.com/Ultimaker/Cura/files/9565879/back.3mf.zip)

GregValiant, 2022-09-14:
Thanks for the report.
There is a persistent bug in 5.1 that keeps certain Hi-Res models from slicing i the model is at the midpoint of the build plate in Cura.  As a workaround, try moving the model on the build plate.  You might have to rotate it as well.
I'll mark this as a duplicate.
Here is your model after I moved and rotated it.
[image]

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release

[image]

closed 2024-01-14 by jellespijker