Ultimaker/Cura #13292: Still having issues slicing certain files since updating CURA

wmerritt72, 2022-09-14:
### Application Version

5.1

### Platform

MacOS 12.6

### Printer

Creality CR-10 Smart

### Reproduction steps

1. import Model and try to print

### Actual results

[image]
[image]

### Expected results

[cura.log](https://github.com/Ultimaker/Cura/files/9569421/cura.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

non

GregValiant, 2022-09-15:
Thanks for the report.
Without a Project File ("File | Save Project") we can't check the model.  Have you checked the failed models for errors?  Are they watertight?

handyrandyrc, 2022-09-16:
I am getting the same error.  Attaching a copy of the project I was working with.
[CE3PRO_Earth_Golem_FINAL.zip](https://github.com/Ultimaker/Cura/files/9584495/CE3PRO_Earth_Golem_FINAL.zip)

handyrandyrc, 2022-09-16:
All good now!  @wmerritt72 - I will bet there is an issue with your model that needs to be repaired.  I opened it up with Microsoft 3D builder, which reported errors with my model.  It was able to repair the errors and I saved it.  I then imported into Cura, and was able to slice without any errors.  There are online tools for doing automated model repair, or you can find an application for your Mac that will do the same.

GregValiant, 2022-09-16:
Thanks @handyrandyrc .
@wmerritt72 This is the MeshMixer view of handyrandy's model.  When a model has this many errors Cura can't figure out how to slice it.  When the "Model is not watertight and may not print correctly" warning comes up you need to take heed.  It doesn't matter how the model was made or who made it or where you got it from, some models have critical errors.
[image]

Here it is repaired.
[image]

And here is the repaired model sliced.
[image]

closed 2022-09-30 by github-actions[bot]