Ultimaker/Cura #13294: Plugin errors

xeorx, 2022-09-15:
### Application Version

5.1

### Platform

Ubuntu 22.04.1 LTS ( 64-bit)

### Printer

Labists ET4

### Reproduction steps

[image]
This plugin errors are allways after start Cura. To remove or new installations doesn't help.

### Actual results

Cura restart.

### Expected results

The same errors.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9571205/cura.log)

fieldOfView, 2022-09-15:
Ultimaker only supports the AppImage they provide for Linux. It seems you are using a different distribution of Cura, which omits one or more dependecies; the trimesh python package is required for both plugins that fail to load.

Can you check the AppImage instead?

xeorx, 2022-09-15:
Ok, with the AppImage works fine. Thx for help.

closed 2022-09-15 by xeorx