Ultimaker/Cura #13296: Cura 5.1.1 modern appimage hangs after initial startup (click the "Get Started" button)

andrigs, 2022-09-15:
### Application Version

5.1.1

### Platform

Fedora 36

### Printer

None yet

### Reproduction steps

1.  Double click on the "modern" version of the app image
2. Splash screen comes up, and the application launches with a dialog and the "Get Started" button
3. Click the "Get Started" button, and it just sits there doing nothing.

I also did this from a terminal window to see if there are any errors, and there are no errors after clicking the "Get Started" button, but here is what is in the terminal window:

[andrigtmiller@andrigsaircooledtechnology Cura]$ ls
Ultimaker-Cura-5.1.1-linux.AppImage  Ultimaker-Cura-5.1.1-linux-modern.AppImage
[andrigtmiller@andrigsaircooledtechnology Cura]$ ./Ultimaker-Cura-5.1.1-linux-modern.AppImage
PyInstaller/loader/pyimod02_importers.py:499: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
Warning: Ignoring WAYLAND_DISPLAY on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
Cyclic dependency detected between "file:///tmp/.mount_Ultimal72nZg/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_Ultimal72nZg/share/cura/resources/qml/Actions.qml"
QQmlEngine::setContextForObject(): Object already has a QQmlContext

### Actual results

Application just sits there doing nothing.  I can close it.

### Expected results

I really don't know, since this is my first time trying to use Cura in years.  It should do something after I click the "Get Started" button, but I'm not sure what.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Nothing to add here.

andrigs, 2022-09-24:
Is anyone ever going to look into this?  I did some tests by going backward from release to release until I found a release that actually worked on Fedora 36.  It turns out that this has been broken for a long time now.  The earliest release that will run on Fedora 36 is the 5.0 Beta+ release.  That's the last one that doesn't hang on initial startup, and will actually work past the "Get Started" button press.

chrta, 2022-09-30:
Have you tried starting it like this:
```
GDK_BACKEND=x11 LD_PRELOAD=/usr/lib64/libstdc++.so.6 ./Ultimaker-Cura-5.1.1-linux.AppImage
```
(see https://github.com/Ultimaker/Cura/issues/12776#issuecomment-1200127332)

I am using this with Cura 5.1.0 in fedora and have not tested any newer version. I have not seen your error message with my failing tests in Cura 5.1.0.

andrigs, 2022-09-30:
> Have you tried starting it like this:
> 
> ```
> GDK_BACKEND=x11 LD_PRELOAD=/usr/lib64/libstdc++.so.6 ./Ultimaker-Cura-5.1.1-linux.AppImage
> ```
> 
> (see [#12776 (comment)](https://github.com/Ultimaker/Cura/issues/12776#issuecomment-1200127332))
> 
> I am using this with Cura 5.1.0 in fedora and have not tested any newer version. I have not seen your error message with my failing tests in Cura 5.1.0.

I had not tried that, but just did, and it does the same thing.  Here is the terminal output:

GDK_BACKEND=x11 LD_PRELOAD=/usr/lib64/libstdc++.so.6 ./Ultimaker-Cura-5.1.1-linux-modern.AppImage
PyInstaller/loader/pyimod02_importers.py:499: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
Warning: Ignoring WAYLAND_DISPLAY on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
Gtk-Message: 07:21:47.239: Failed to load module "canberra-gtk-module"
Gtk-Message: 07:21:47.239: Failed to load module "pk-gtk-module"
Gtk-Message: 07:21:47.240: Failed to load module "canberra-gtk-module"
Gtk-Message: 07:21:47.240: Failed to load module "pk-gtk-module"
Cyclic dependency detected between "file:///tmp/.mount_Ultima5Iuum1/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_Ultima5Iuum1/share/cura/resources/qml/Actions.qml"
QQmlEngine::setContextForObject(): Object already has a QQmlContext

andrigs, 2022-10-18:
Strangely enough, I tried to run this again, without any parameters, and just double clicked the app image.  If I wait long enough, it actually goes past the "Get Started" button press, and allows me to add my printer, and my filaments, and slice a model.  The UI performance is very poor, but generally the software is working on Fedora 36.  Not sure why the UI performance is as poor as it is, but it's working.

closed 2022-10-18 by andrigs

faern, 2023-04-16:
> If I wait long enough, it actually goes past the "Get Started" button press

How long is "long enough"? Because I have a very similar issue right now, and it's been stuck for many minutes. Not at the startup screen per se. But any time I'm trying to add a printer the entire program locks up completely somewhere random in the add-printer wizard.

andrigs, 2023-04-16:
Well, it was quite a long time from the initial screen.  I'm not sure how
long it was because I walked away and did some other things, and came back
and suddenly it was on the next screen.  I think it was some 10 minutes
that I was away, and maybe even a bit longer.

Andrig
President, Andrig's Aircooled Technology
Confidential