Ultimaker/Cura #13301: Does not open folder after file save

cropduster32, 2022-09-16:
### Application Version

4.13.2

### Platform

Dell, Ubuntu 20.04.5 LTS, 64-bit, Gnome 3.36.8, X11

### Printer

Creality 10S

### Reproduction steps

1. After opening 3 separate items onto the workspace and separating them out, hit slice and saved the result to a file.
2. A window popped up to show the saved file with an option to open folder where the file was saved.
3. The Open Folder option will not work

### Actual results

Nothing happens, the folder is not opened and there are no messages to indicate why a folder is not opened.

### Expected results

I was expecting a folder to be opened so I could copy the file to a card file system for printing.
[image]
[cura-sep16-11-54.zip](https://github.com/Ultimaker/Cura/files/9583426/cura-sep16-11-54.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[Clew adjuster 11mm v2-compressed.zip](https://github.com/Ultimaker/Cura/files/9583443/Clew.adjuster.11mm.v2-compressed.zip)

cropduster32, 2022-09-16:
Update - Cura is loaded and maintained by snapd

nallath, 2022-09-16:
We don't formally support anything but the Appimage version that we provide. Have you tried if it works there? 

Are you also sure that you're using 4.13.2? We never released that version.

cropduster32, 2022-09-16:
[image]
The version I have is maintained with Snap so I was expecting the latest version.
I'll try the AppImage and get back to you.

cropduster32, 2022-09-16:
Yes, the 5.1.1 AppImage works as I'd expect. So why does Snap maintain an old version?

nallath, 2022-09-16:
> Yes, the 5.1.1 AppImage works as I'd expect. So why does Snap maintain an old version?

I don't know. I also don't know who maintains that version.

closed 2022-09-16 by nallath