Ultimaker/Cura #13302: Hangs on "Initializing Package Manager...."

silvrubi592, 2022-09-16:
### Application Version

5.1.1

### Platform

Windows 10/11

### Printer

Creality Ender 3 Pro

### Reproduction steps

1. I clicked the icon to open Cura.
2. Cura started to open and then stops on "Initializing Package Manager...."

### Actual results

NOTHING!!!! THE PROGRAM NO LONGER OPENS!!!!

### Expected results

THE PROGRAM SHOULD OPEN SO I CAN IMPORT AN OBJECT AND SLICE IT

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Windows may have run an update

MariMakes, 2022-09-28:
Hey @silvrubi592,

I'm sorry to hear you are stuck! Sorry, it took us a while to get back to you. 😞 
Some fresh logs would really help to troubleshoot.

You can find your log file here:
Windows: %APPDATA%\cura\<Cura version>\cura.log or usually C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log

In this case, your configuration folder could also be helpful. 
You can share your configuration folder by:
	1. Go to Help at the top bar of Cura
	2. Select Show Configuration Folder
	3. Select all the folders in there and create a zip OR go up a folder in your folder structure and zip your folder with the correct Cura version
You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser. 

We have some known issues that are related to having an extra monitor starting from Cura 5.x Do you have an extra monitor connected that might influence your Cura? 🤔

closed 2022-10-20 by github-actions[bot]

akenf, 2022-11-17:
Hello,

i have the same Problem with cura version 5.2.0 and 5.2.1.

Steps:
- installed cura 5.2.0. It opened. i closed it and tried opening it 10 min later. It gets stuck on "initializing package manager". Nothing moves in that window.
- deinstalled it and installed 5.2.1. Gets stuck again. Didnt work once.
- deinstalled it and installed 5.2.0 again. Didnt work. it gets stuck.
- deinstalled and installed in "C:/Cura..." (someone sugested online). Doesnt work.
- also tried to deactivate windows firewall. doesnt help

I have a windows surface pro 4 laptop with windows 10. No extra monitors connected.

What extra info would you need?
Thank you.