Ultimaker/Cura #13303: "Enable Prime Tower" Checkbox unvisible

Wolfi12345, 2022-09-16:
### Application Version

5.1.0

### Platform

Windows 10 64bit Deutsch

### Printer

Ultimakter 3

### Reproduction steps

1. Visibility Settings --> ALL
2. Diselect support

### Actual results

1. Prime Tower not visible --> correct
2. BRIM of the Prime Tower is visible
3. Try to uncheck "Prime Tower enable", but the Line doesn´t appear anyway. 

### Expected results

If Setting visiblity to ALL, the line with the "Prime Tower enable" should appear

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]

MariMakes, 2022-10-04:
Hey @Wolfi12345,

Thanks for the report 👍 
And welcome to the Ultimaker Cura Github 🚀 
Sorry it took us a while to get back to you. 😞 

I think I understand what is happening. 
The primetower setting is not visible in single extrusion.
And since you've disabled your second extruder, the setting is hidden.
But, I believe that in your case it might have been enabled, perhaps by accident. 
If you enable the second extruder the setting re-appears and you can disable it there.

https://user-images.githubusercontent.com/40423138/193845445-9b4e1970-9c17-42d7-982e-5749df8f25c0.mp4

I'll ask the team why the setting is not disabled when it's hidden. 
Can you let me know if this workaround works for you? 
Thank you 😄

MariMakes, 2022-10-06:
We are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs. We believe we will not work on this anytime soon and will therefore defer this issue.

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.

closed 2022-10-06 by MariMakes