Ultimaker/Cura #13305: Bug report or feature request? Files added are always off build plate. Should Always offset part so it's on plate

tastyratz, 2022-09-16:
### Application Version

5.1.1

### Platform

Windows 10

### Printer

Monoprice Maker ultimate

### Reproduction steps

When dragging a file into the cura window to add a part it is always added off the build plate in an unprintable location.

[image]

### Actual results

Cura alignment always puts 0,0 of the part with 0,0 of the build plate. This means it will never actually drop in a printable way.

### Expected results

Cura should center the part on the build plate as if the machine was set for origin at center even if the printer does not have an origin at center configuration.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Second best would be to at least establish manifold furthest edges when cura detects a part is off the build plate and offset the location by the amount it's off manifold to fit. This might even make a good second-step automatic operation so more parts dropped on a plate are printable as dropped. 
Other ideas could be to automatically run an "arrange all models" operation only on new parts dropped onto the plate or perhaps also constrained to both part and the axis which does not fit.

This has been the case for a very long time. While it's always been just a minor annoyance, I would imagine part start as if machine origin is center regardless of machine setting could be an easy user-friendly code change.

I am not sure if this is a bug because this seems like what SHOULD already be happening or if it's a feature suggestion. The idea is to make the print a little more drag and drop with less manual manipulation for ease.

GregValiant, 2022-09-16:
Thanks for the report.
My STL files always come in with the part Center of Geometry located at the mid-point of the Cura build plate.  Generally, there is no location information in an STL file.
3mf model files can contain location information and they will sometimes come in at the build plate origin which is the case in your image and yes, in those cases the model must be moved as it is in a non-printable location.  It is where the 3mf told Cura to put it though.
If you are generating the files in a CAD program and exporting them then I wonder if there is an option in the export utility you are using so location information is not contained within the exported file.
I will leave this as a bug report and someone from the Cura team will take a look.  In the meantime if you would explain what type of files you are opening in Cura and how they are being generated that would be helpful.

tastyratz, 2022-09-17:
You know what? That’s a good point. The example in my picture was a VERY basic peg insert made in windows 3d builder. I am just realizing from your comment that this behavior is happening with 3mf files. I tested several of them and had the same thing happen while STL files do not share the same behavior.

Should Cura be using this location information for part positioning, however? Or should it be treating 3mf files in the same way as STL files? After all, that location and point of origin shouldn’t be relevant to build plate positioning and does not seem to be taken into account with STL files?

GregValiant, 2022-09-17:
The location information can be important when parts are going to be merged.  Because of that Cura would have to be tasked with making decisions I think would be better left with the user.
In 3D Builder, you can save your model as an STL file and it will come into Cura at the midpoint of the build plate.

I don't see a bug here as for both 3mf and STL it is expected behavior.  I'll remove the bug tag on this and leave it open so you can respond.  Maybe you would want to close this ticket and open a feature request for an "Insertion Point" option?

tastyratz, 2022-09-17:
When you say merged, are you talking about assemblies or people building parts in multiple files to print as 1 part in Cura? I'm trying to understand the use case and it sounds very specific.

In any instance, if the location information generates a part that is placed in an unprintable location (such as the example above) does that location information still have value to be retained? 
In the instance of multiple files being imported that have that location information, maybe it instead makes sense to retain a positional offset when adjusting the position for the parts to actually fit on the plate. After all, the idea is to allow a more simple print for the most common use case but the offset can be attached to the object's x, y coordinates in Cura should someone actually want to perform that function so as to not reflect 0/0/0 in object properties.

GregValiant, 2022-09-17:
The primary use of "Merge" is for printing multi-color prints.  Each model within the merge can be set to a different extruder.  The Merge function puts the parts together as they should be.
Cura is not an assembly modeler.  It can be tricked into properly assembling parts but it's a slicing program and although you can do things with support blockers and such, Cura has no real CAD abilities.

We are only talking about 3mf vs STL but Cura supports other model file types as well.  I think the feature request would need an understanding in regards to those other file types (OBJ, GIF, PLY, etc.) as they would also be affected.
Think about it and maybe you can write something up.  It isn't only the Cura team at UltiMaker but there are other developers out there who might pick up the ball and run with it.
Not me though.  I'm just the greeter here.

tastyratz, 2022-09-22:
So, I've been mulling on this a little.

It sounds like the real problem is around relative positioning in respect to multiple files being imported. If we add part A, we want part B to land in the same spot.

What I'm still struggling with here is why using a center reference point for 0,0 over a corner reference point of 0,0 in imported parts is any different considering some printers may be explicitly configured to use the center point as 0,0 in machine options. This seems like the most logical step for approaching files with positional data in them (like 3mf) or are we worried about the use case for individuals who are utilizing 3mf file positioning as build plate positioning?

I feel like the most likely scenario here is exactly how I used it, windows 3d builder created basic files and dropped onto the plate which would gain from the increased user friendly handling.

If the build plate center point is considered 0,0 for part positioning, this would still behave exactly the same in that parts that need to be merged would have a shared relative origin. Those parts (on a 200mm plate) can either be represented as 0,0 or 100,100 after import where they land and as long as they have a shared origin then I'm not sure why that wouldn't work just the same?

What I'm thinking we avoid here is that there are instances cura can add a part that automatically starts off the build plate

tastyratz, 2022-09-23:
Truthfully, it's not a massive inconvenience and I am able to just rearrange plate, It's not it just seemed like a way to reduce steps for the scenario and like an easy win code-wise since Cura already knows how to handle machines with a 0,0 center origin under machine settings. STL files are imported with a center origin while 3mf is not. It would just be importing ALL files treating the center point as origin. 

If not, this can be worked around as a minor annoyance that has an additional step. It isn't showstopping. Maybe I underestimated any complexity.

GregValiant, 2026-09-10:
I'll go ahead and close this.

closed 2026-09-10 by GregValiant