Ultimaker/Cura #13306: slice failed

djwest9999, 2022-09-16:
### Application Version

5.0.0

### Platform

windows 10

### Printer

ender 3 pro

### Reproduction steps

clicked on slice of a jeep duck and failed

### Actual results

wouldn't slice

### Expected results

print 3d image 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

just reporting it

GregValiant, 2022-09-16:
@djwest9999 all these reports are taken seriously by the developers but there can be no investigation without a Project File and a Cura.log file
In addition, if you are trying to print an image then there have been a couple of other reports about problems with that.  More information gives the team a better chance of chasing down a bug.

djwest9999, 2022-10-11:
Due to the time it is taken to respond I have actually figured it out
myself. Your software is still the the only one I use. Thank you anyway for
your help

MariMakes, 2022-10-11:
Thanks for the update 👍 
I'm happy to hear you've been able to figure it out. 😊

closed 2022-10-11 by MariMakes