Ultimaker/Cura #13308: Missing Tree Support structure

DragonDon, 2022-09-16:
### Application Version

5.1.0

### Platform

MacOS 10.15.7

### Printer

Creality Ender 5 Plus

### Reproduction steps

1. Load File
2. Slice file

### Actual results

I havd a model with 7 holes.  The tree support only go to 6 of the holes.  I have no restrictions on where supports are placed.

[image]

### Expected results

I was expecting all 7 holes to have supports generated for them.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[LED Plane V8.zip](https://github.com/Ultimaker/Cura/files/9587010/LED.Plane.V8.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/9587018/cura.log)

GregValiant, 2022-09-16:
Thanks for the report.
This is with my normal everyday settings profile.
[image]

To the right of the Settings Search box is the Settings Visibility tool.  Click on it and set the visibility to "All".  All 73 Support settings should be visible.  You need to configure them so they will work with this model.

If it was mine, I'd use normal supports instead of Tree supports.  The supports would only be in the holes.
[image]

I think we have a settings issue here rather than a bug so I'll remove the bug label and I'll leave this open so you can respond.

DerGenaue, 2022-09-25:
Cura's current tree support implementation sometimes loses branches.
You can try moving the model around on your buildplate, rotating it and changing values such as `Tree Support Branch Angle` to find a configuration where all branches generate correctly.

Alternatively, you can also try out this [alternative tree support implementation](https://github.com/ThomasRahm/CuraEngine) which should fix those issues.

DerGenaue, 2023-07-04:
Please check if this has been solved in the new Cura 5.4 release,
It features a new tree support implementation

DragonDon, 2023-07-05:
Hey Daniel,Yes, the supports are there.  Thanks for the follow-up!Don