Ultimaker/Cura #13309: Slicing failed with an unexpected error. Please consider reporting a bug on our issue tracker (Backend exited abnormally with return code 3221225477)

Loki44, 2022-09-16:
### Application Version

5.1.0

### Platform

Windows 10, intel UHD graphics 600

### Printer

ultimaker 2+

### Reproduction steps

try to slice
get error telling me to come here.

### Actual results

I got told to report a bug

### Expected results

the model should slice.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

I noticed "Backend exited abnormally with return code 3221225477" in the log, I'm guessing that is why the error was thrown.

[UM2_CR-10_HD-modular-mount-V6_mount-body__volcano_v1.2.zip](https://github.com/Ultimaker/Cura/files/9590406/UM2_CR-10_HD-modular-mount-V6_mount-body__volcano_v1.2.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/9590408/cura.log)

GregValiant, 2022-09-16:
Thanks for the report.
I am unable to duplicate the problem.  I opened the 3mf and it sliced fine.  (It does look like your supports could use some support though.)
The "3221225477!" error code has been coming up in other bug reports so I'll leave this open and mark it as a duplicate.
You may be able to get this to slice by grouping the models and moving/rotating the group on the build plate.  That workaround has helped on a lot of these.

Loki44, 2022-12-15:
Thanks, I haven't had this issue with 5.2 so I assume it's fixed.

closed 2022-12-15 by Loki44