Ultimaker/Cura #13311: Failed to slice big sphere model

TheHackerArtist, 2022-09-17:
### Application Version

5.1.1

### Platform

Windows 10 Pro

### Printer

Custom Printer

### Reproduction steps

1. Loaded a large sphere model
2. Sliced Using cura
3.Find attached log file:
[cura.log](https://github.com/Ultimaker/Cura/files/9590881/cura.log)
Also the model was a 40Mb STL files
[image]

Also link to .zip for project file
[CFFFP_1692 hexa.zip](https://github.com/Ultimaker/Cura/files/9590892/CFFFP_1692.hexa.zip)

### Actual results

As seen in attached image there was an error with slicing

### Expected results

Should have sliced model

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

See other sections for links to files

GregValiant, 2022-09-17:
Thanks for the report.
MS 3D Builder didn't find ay errors in the model.  I simplified the model to a 3mb file and it still wouldn't slice in 5.1.
None of my usual tricks worked.  I uploaded the simplified model to a [repair site](https://formware.co/OnlineStlRepair) (it has a 20mb file size limit).  It didn't find any errors but made some changes:
-> Vertex count changed from 31784 to 91732 (+59948)
-> Triangle count changed from 70324 to 217908 (+147584)
The altered file is 10mb.
Next, I deleted your model from Cura and opened the fixed file.  Cura worked on it for about 40 minutes and was able to successfully slice the model.
Cura 4.13 was able to slice the original 40mb model.
This behavior (won't slice a model that has no errors) has been noted before.  I'll mark this as a duplicate.  Someone from the Cura team will take a look and see if this is indeed a duplicate or if this model is causing a different problem.

TheHackerArtist, 2022-09-17:
Thanks for the reply & follow up!

Glad you've hit a new GCode size milestone! haha

> Cura 5.1 worked on it for about 40 minutes

Out of curiosity, what are the system specs you were running this on? Any idea why S3D took about 3 minutes to slice the 40mb file vs 40 min on Cura for your updated 10mb file? 

I'll often find myself having to slice similarly large files in the upcoming months.

Cheers!

Piezoid, 2022-09-17:
I can confirm that this is another case of invalid trapezoidal graph. It segfaults [here](https://github.com/Ultimaker/CuraEngine/blob/7b802278b8bc3218e7af92384009d7734a94999d/src/SkeletalTrapezoidationGraph.cpp#L316).
<details>
<summary>stack trace</summary>

```
    #0 0x555b99ea8eb8 in cura::SkeletalTrapezoidationGraph::collapseSmallEdges(long long) CuraEngine/src/SkeletalTrapezoidationGraph.cpp:316:34
    #1 0x555b99e57e85 in cura::SkeletalTrapezoidation::constructFromPolygons(cura::Polygons const&) CuraEngine/src/SkeletalTrapezoidation.cpp:454:11
    #2 0x555b99e57c21 in cura::SkeletalTrapezoidation::SkeletalTrapezoidation(cura::Polygons const&, cura::BeadingStrategy const&, cura::AngleRadians, long long, long long, long long, long long) CuraEngine/src/SkeletalTrapezoidation.cpp:376:5
    #3 0x555b99db96eb in cura::WallToolPaths::generate() CuraEngine/src/WallToolPaths.cpp:105:28
    #4 0x555b99eeeb05 in cura::WallsComputation::generateWalls(cura::SliceLayerPart*) CuraEngine/src/WallsComputation.cpp:72:48
    #5 0x555b99eef83c in cura::WallsComputation::generateWalls(cura::SliceLayer*) CuraEngine/src/WallsComputation.cpp:88:9
    #6 0x555b99e3d54b in cura::FffPolygonGenerator::processWalls(cura::SliceMeshStorage&, unsigned long) CuraEngine/src/FffPolygonGenerator.cpp:714:23
    #7 0x555b99e3d54b in cura::FffPolygonGenerator::processBasicWallsSkinInfill(cura::SliceDataStorage&, unsigned long, std::vector<unsigned long, std::allocator<unsigned long> > const&, cura::ProgressStageEstimator&)::$_3::operator()(unsigned long) const CuraEngine/src/FffPolygonGenerator.cpp:492:36
```
</details>

Disabling mesh vertex merging didn't help in this case. My branch with integer resolution increased from 1 µm to 0.01 µm *and* floating point calculation could slice the original 3mf. However increased resolution alone wasn't enough, floating point helped even if there weren't overflows, for some reason. :thinking: 
It generated a 1018 MB gcode file with 26 million lines!

> Out of curiosity, what are the system specs you were running this on? Any idea why S3D took about 3 minutes to slice the 40mb file vs 40 min on Cura for your updated 10mb file?

S3D has "Simplify" in the name for a reason, it has much simpler algorithms and it must use a more severe simplification.
This mesh has fine filters on the edges, and a high resolution on the sphere. 

BTW, the cell walls are too fine at their base for the 0.8mm line width:
[image]

GregValiant, 2022-09-17:
I run Cura on an old tired laptop (not unlike it's owner).  The video subsystem is poor and likely why both 4.13.1 and 5.1 crashed when attempting to display the gcode.
But enough is enough.  Monday I'm going to order a capable machine and replace this one.

jellespijker, 2024-01-14:
Thank you for reporting this issue, I can confirm that the project file still crashes with our latest internal build 5.7.0-alpha.0+3abb55 . For internal reference the crash is logged crash under Sentry issue:
[CURAENGINE-1Y](https://ultimaker-o7.sentry.io/issues/4863562093/)