Ultimaker/Cura #13313: "Pause at height" extension does not pause at right height

Denibel, 2022-09-17:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Creality Ender 5

### Reproduction steps

On a very simple piece (4 layers only, each 0,2 mm). Ask for "pause at height" at 0,6 mm (end of layer 2, before starting layer 3), examine G-code.

### Actual results

Pause is inserted in G-code at the end of layer 1, before starting layer 2, i.e. at a height of 0,4 instead of 0,6.

### Expected results

Pause after finishing layer at 0,4 height, before starting layer at height 0,6

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

https://github.com/Ultimaker/Cura/files/9592208/Pause.at.height.test.zip

GregValiant, 2022-09-17:
Thanks for the report.
I always use the "By Layer" function so I've never noticed this.  Using "By Height" I get the same results as you did with the print pausing at Z=0.6.
The result is that
LAYER:0  Z=0.2
LAYER:1 Z=0.6
LAYER:2 Z=0.6
LAYER:3 Z=0.8
Someone from the Cura team will look at this.

pnt103, 2022-12-01:
@GregValiant  See [#9430](https://github.com/Ultimaker/Cura/issues/9430).

I suspect this is a bug caused by the script reading the Z value from the wrong place.  In very old versions of Cura, the gcode to move the nozzle to the "next" layer and make that the "current" layer (from the point of view of the script) was at the start of the layer code, ie after the ";LAYER:_xx_" comment.  Consequently the pause script looked at the start of the "current" layer - in this case the layer prefixed by ";LAYER:1" - and read ahead to find the starting X, Y and Z coordinates for that layer.  Some time later, the gcode layout changed such that the move to the start position for the new current layer was placed at the end of the previous layer. Therefore when the script read the first X and Y coordinates it found, they were actually the end of the first extrusion rather than the start.  Similarly, it took the first Z value it could find by reading ahead, but that was actually the Z value from the move to the next (not current) layer.

I thought I'd fixed that, and my changes were incorporated in Cura 4.9 but apparently I didn't quite get it complete for pause at a height rather than pause at a layer.

GregValiant, 2023-01-09:
I missed this reply.
I made some changes to Pause at Height and submitted them.  We'll see.  I'm not savvy enough at Python to tackle this particular issue though.
Here is my effort on the script.  The basic changes were to:

- Re-order the prime so it occurs before moving back to the print which eliminates a blob problem.
- Changing some tooltips
- Adding an M104 option for the "resume" to eliminate what is often a 10 second delay for no reason.
- I added multi-line capability to "Gcode before pause" and "Gcode after pause".
[GV_PauseAtHeight.zip](https://github.com/Ultimaker/Cura/files/10369876/GV_PauseAtHeight.zip)

Denibel, 2023-03-04:
At last I tried your version with Cura 5.2.1. Pause at layer works properly : stops when whanted, resumes to proper height (not a suspended layer in the air !). Thanks !
I do not tried pause at height.

GregValiant, 2023-03-05:
I don't use "at Height" because that other issue aside - Z-hops confuse it as the Z is bouncing up and down, and with Adaptive Layers the height is not consistent.  "By layer" has always worked for me.