Ultimaker/Cura #13315: Unexpected error hit when trying to slice

astead, 2022-09-17:
### Application Version

5.1.0

### Platform

Windows 10 

### Printer

Creality Ender-3 Pro

### Reproduction steps

Load an STL and then hit slice.

### Actual results

Get error message:
Slicing failed
Slicing failed with an unexpected error. Please consider reporting a bug on our issue tracker.

### Expected results

Finish slicing the model and create gcode.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9592365/cura.log)
[CE3PRO_yos_a3.zip](https://github.com/Ultimaker/Cura/files/9592369/CE3PRO_yos_a3.zip)

GregValiant, 2022-09-17:
Thanks for the report.
This bug has been reported numerous times so I'll mark this as a duplicate.  The model appears to be free of errors.
You can try simplifying the model in a program like MS 3D Builder.  After simplifying (I used the "2" factor) the model did slice.
[image]

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release

[image]

closed 2024-01-14 by jellespijker