Ultimaker/Cura #13316: Do not change the settings in the intersection. Can't change fill options at intersection

komp0t2, 2022-09-17:
### Application Version

5.1.1

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

1.We take a model. 
2.We put the block of removal of supports. 
3.We change the settings.
 Nothing happens

### Actual results

Nothing happens

### Expected results

Change the fill settings in the intersection area

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

There is nothing to add. does not work on any model

Asterchades, 2022-09-18:
You'll need to add at least a project file for anything to be done about this. It's working fine for me, which means it's likely to be working fine for the people who would be looking into what's wrong with it as well. If it cannot be reproduced, it cannot be fixed.

[image]
(3D Benchy with a chunk being modified from 10% infill to 50%)

komp0t2, 2022-09-18:
This works with a smaller padding value. like in your case.
In my case, a transition from 100 percent filling is necessary
[image]
I also tried any other settings at 100% infill. Like the amount of wall
[image]

For version 4.13 this worked fine