Ultimaker/Cura #13317: Post Processing "Pause at Height" adds extra extrusion command which causes a glob of filament on top of the print before continuing

derekantrican, 2022-09-17:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

I sliced a model to practice with two-color printing. Following some online articles, I applied the post processing script for "Pause at Height" and used the following settings:

[image]

Here is the code that was inserted:

```
;TYPE:CUSTOM
;added code by post processing
;script: PauseAtHeight.py
;current layer: 10
M83 ; switch to relative E values for any needed retraction
G1 F3000 E-125.0
G1 F300 Z3.2 ; move up a millimeter to get out of the way
G1 F9000 X20 Y20
G1 F300 Z15 ; too close to bed--move to at least 15mm
M104 S200 ; standby temperature
M117 Change Filament
M0 ; Do the actual pause
M109 S200 ; resume temperature
G1 F300 Z2.2
G1 F9000 X115.218 Y116.811
G1 F300 Z2.2 ; move back down to resume height
G1 F3000 E125.0
G1 F2700 ; restore extrusion feedrate
M82 ; switch back to absolute E values
G92 E277.89589
```

For some reason, it seems that the post processing script is programmed - upon resuming - to move the printer back to the top of the print, then runs an extrude command (`G1 F3000 E125.0`). In my testing, this means that the nozzle moves back to the printer, then just squirts out a glob of filament right on top of the print - not great for results. Removing this line before sending to the printer gave me a much cleaner layer on top.

As you can see in my screenshot, I had "Extrude Amount" set to 0 (through testing, I found setting this to something else actually adds _more_ extrude commands) so I'm not sure what the `E125` line is for. Why does this line exist?

### Actual results

Extrude command inserted for printer resuming, causing glob of filament in next layer

### Expected results

Expected the printer to resume printing like normal

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

N/A

druckgott, 2022-09-18:
I think the Problem is:
retraction_amount and extrude_amount

`
                    # Optionally extrude material
                    if extrude_amount != 0:
                        prepend_gcode += self.putValue(G = 1, E = extrude_amount, F = 200) + "; Extra extrude after the unpause\n"
                        prepend_gcode += self.putValue("@info wait for cleaning nozzle from previous filament") + "\n"
                        prepend_gcode += self.putValue("@pause remove the waste filament from parking area and press continue printing") + "\n"

                    # and retract again, the properly primes the nozzle when changing filament.
                    if retraction_amount != 0:
                        prepend_gcode += self.putValue(G = 1, E = -retraction_amount, F = 6000) + "\n"
`

If you have extrude_amount == 0 it will happen nothing so this lines will not be added
But if retraction_amount != 0 then it will add a -retraction

The question is how it makes sence to handle it.

Maybe the -retraction makes no sence because its no retraction anymore its more an extrude or maybe pre_extrude.
At first I think it´s needed to specify the goal?

GregValiant, 2022-09-18:
@derekantrican thanks for the report.
@druckgott thanks for the analysis.
Here is my 2 cents.
Up until the point that you release the pressure wheel of the extruder, Cura knows exactly where the end of the filament is.
Then you remove the filament and insert the new one.
Now there are three possibilities.

1. The end of the filament is _exactly_ 125mm from the end of the nozzle so it is right where Cura thinks it is.  That is unlikely in the extreme.
2. The filament was inserted so the end is more than 125mm from the end of the nozzle.  The nozzle will not extrude any plastic until the filament takes up the gap.  The print will resume severely under-extruded.
3. The filament end is less than 125mm from the nozzle.  You will get a larger or smaller prime blob that is at retract speed, not extruding speed.  I would think there would be a distinct possibility of missed steps.  This prime blob is timed to happen over the print.

Once the pressure on the filament is released and the filament moves even a little bit then Cura no longer knows where the end of the filament is in relation to the nozzle.

- I agree that advancing the filament "retraction amount" after the pause is a bad idea.  Of the three things that can happen - two of them are bad.  The only upside is that Cura once again knows exactly where the end of the filament is.
- I also agree that even if the user asks for a prime-after-pause it should be done before moving back over the print.
Using the 125mm from the example - it is up to the user to get the filament end located to exactly 125mm from the nozzle.  That likely never happens.

If the order of the commands was changed to:
M0 ; Do the actual pause
M109 S205 ; resume temperature
**G1 F2100 E125 ;>>advance the filament**
G1 F2100 ; restore extrusion feedrate
G1 F300 Z5.2
G1 F9000 X76.569 Y130.718 ;**>>>Move to the restart location**
G1 F300 Z5.2 ; move back down to resume height

Whether or not there should be a prime seems to be covered in DruckGott's analysis.  When that prime occurs appears to be a second issue.
I use PauseAtHeight a lot.  It is my experience that not enabling the retraction works best.  When I insert the new filament I push it by hand to purge the previous color (going from any color to white takes a lot of filament).  After purging, I do a hand retraction of 3 or 4mm and then resume the print.  The best "resume" situation is if the print restarts within support.

Someone from the Cura team will take a look at this.

derekantrican, 2022-09-21:
Thanks for the analysis. It sounds like the assumption is that the same distance you retract at the beginning of the pause, should also be the same distance extruded after the pause. @GregValiant similar to your usage of PauseAtHeight, I "purge" the previous filament by hand. I do not, however, do a hand retraction before resuming the print (though maybe I should).

I do agree that if there should be a "post-pause extrusion" at all, it should happen _before_ the nozzle moves back to the print.

a5421306, 2024-08-18:
Can you give me a copy of the plugin

GregValiant, 2024-08-18:
The behavior was changed not long after the above problem was noticed.  Any extra extrusion now takes place at the park position.
If you have a recent version of Cura (5.8.0 seems very good) it will have the updated "Pause at Height" script.
If you are having other problems you need to open a bug report ticket.

closed 2024-08-18 by GregValiant