Ultimaker/Cura #13318: Menu items in window instead of menu bar on Mac OS

OutcastZeroOne, 2022-09-18:
### Application Version

5.1.1

### Platform

Mac

### Printer

all

### Reproduction steps

launch program.

### Actual results

menu shows on window, not on menu bar like every other Mac OS program.

### Expected results

a program that conforms to Mac OS standard.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

fieldOfView, 2022-09-18:
This has been reported before, see https://github.com/Ultimaker/Cura/issues/12307. Therefore I will close this issue as a duplicate.

closed 2022-09-18 by fieldOfView

OutcastZeroOne, 2022-09-18:
well, it has been months and no fix on it yet. not using the new CURA till it works correctly.