Ultimaker/Cura #13320: Problems on Creality CR6SE and current version

Deifel4096, 2022-09-18:
### Application Version

5.0.0

### Platform

MAC OS

### Printer

Creality CR 6SE

### Reproduction steps

Sliced a STL file with the current 5.0.0 version and printed it with the Creality CR 6SE printer.

### Actual results

The outer and inner walls on 2 sides of the printed models have problems (see picture).
The same model does NOT have problems when sliced with an older CURA version 4.13.1.

### Expected results

correctly printed walls.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Project_on_Cura5.3mf.zip](https://github.com/Ultimaker/Cura/files/9593520/Project_on_Cura5.3mf.zip)
[image]
[image]

GregValiant, 2022-09-18:
Thanks for the report.
I sliced your 3mf file and analyzed the code in MS Excel. All the lines in that area of the print have extrusions and the flow of those extrusions varies between 100.03% and 99.85%.  That's a pretty narrow range and the variance is acceptable.
I rotated the model so I could watch it print and then printed some of the model (on my Ender 3 Pro) including the area of interest.  I can see a couple of dimples where the infill touches the walls (see the yellow arrows in the image).  The rest of the model printed well.  The thin area corresponds to the wall section where you show the defect.  It is the first wall extrusion after the z seam.  I'm wondering if the pressure in the nozzle is falling off due to all the tiny infill dots and then the long travel move back to the start point.
Could you post the gcode file that printed that flawed part?  

[image]

Deifel4096, 2022-09-19:
[Card_Box_v2.gcode.zip](https://github.com/Ultimaker/Cura/files/9599915/Card_Box_v2.gcode.zip)

And here the GCODE

GregValiant, 2022-09-20:
Your gcode looks a lot like the one I came up with.  The flow numbers look good and vary between 100.50% and 99.50% for the layers where the half circle exists.  So the problem doesn't appear to be in the gcode.
There is a post-processor named Retract Continue that spreads a retraction across the following travel move.  That can help to bring the nozzle pressure back up after a long combing moves like in that print.
In 4.13 and earlier versions the default wall order with Inside to Outside.  Cura 5.1 has a new setting "Wall Order" and the default was made Outside to Inside.  Make sure you have it set to "Inside to Outside".  It won't cure what is going on but it will hide the under-extrusion within the inside wall.
High print temperatures can cause oozing during combing moves which can lead to your problem.  But your 190° print temperature is at the bottom end of the scale and is 20° cooler than I print at.  Have you ever run a PID Auto-tune on that hot end?  You need Pronterface  to run it because you need to see the results so you can pass the new numbers to the printer.
The command is M303 C8 S200 and I would suggest running it with the layer cooling blower on at 100%.  Even if the PID numbers are near what you have now running it would fall under "Can't hurt".
I can't decide if this is a bug or not.  Someone from the Cura team will take a look.  Since I'm undecided I'll leave it as a bug report but I think the problem is a little bit Cura settings and a little bit how your printer is reacting.

Deifel4096, 2022-09-20:
> Your gcode looks a lot like the one I came up with. The flow numbers look good and vary between 100.50% and 99.50% for the layers where the half circle exists. So the problem doesn't appear to be in [...]

Many thanks for your great investigation and support :-)
I will first try it out with the newest Cura 5.1 version. I have also posted my problems in a Facebook group and someone had the same problems which seems to be solved with 5.1. for him.