Ultimaker/Cura #13321: I cannot open Cura 5 on Mac OS 10.15.7

mariobrussi, 2022-09-18:
I also have this problem

I also cannot open Cura Slicer. I am running Mac OS 10.15.7 which is the latest version of MacOS available for by Macbook Pro 2012.

_Originally posted by @Darrenmeehan in https://github.com/Ultimaker/Cura/issues/12505#issuecomment-1250230006_

Darrenmeehan, 2022-09-18:
I did some digging and I can run all 5.x releases except 5.1.1. I'm using 5.1.0 at the moment and it opens fine.

When I open 5.1.1 I get a prompt warning me it was downloaded from the internet, and it the app's icon bounces once, then nothing. I wonder was it signed correctly? As there's very few commits between 5.1.0 & 5.1.1.
I'll do some more investigation into 5.1.1 to see if I can find out anything else.

Darrenmeehan, 2022-09-18:
I can see the following log in `Console` when Cura 5.1.1 is launched.

```
Error loading Python lib '/Applications/Ultimaker-Cura.app/Contents/MacOS/libpython3.10.dylib': dlopen: dlopen(/Applications/Ultimaker-Cura.app/Contents/MacOS/libpython3.10.dylib, 10): Symbol not found: _preadv
  Referenced from: /Applications/Ultimaker-Cura.app/Contents/MacOS/libpython3.10.dylib (which was built for Mac OS X 11.6)
  Expected in: /usr/lib/libSystem.B.dylib
 in /Applications/Ultimaker-Cura.app/Contents/MacOS/libpython3.10.dylib
```

mariobrussi, 2022-09-20:
Thanks for your insight @Darrenmeehan , i'll keep using 5.1 in the meantime :)

dhysuiej, 2022-10-05:
I was able to run the Cura 5.1.1 on macOS Catalina (10.15.7) with these two files copied from Python 3.10 installed with Homebrew

- libpython3.10.dylib
- _ctypes.cpython-310-darwin.so

mariobrussi, 2022-12-11:
for anyone who's looking for a way to use newer cura (higher than 5.1) on Mac OS Catalina (10.15.7), they've decided to not support this os anymore, this makes me kinda upset as seen as a slicer shouldnt need a new computer to work.
Off course using an outdated version, and more importantly a not upgradable one, doesnt have a sense for anyone except whoever found his forever workflow, i'll turn back to prusaslicer, a little angrily also if i can say.

closed 2022-12-11 by mariobrussi