Ultimaker/Cura #13323: Random bogus defaults in various printer profiles

richfelker, 2022-09-18:
### Application Version

5.1

### Platform

any

### Printer

lots

### Reproduction steps

Select a printer-specific profile shipped with Cura

### Actual results

Find a large number of settings set to dubious (often snakeoil or even outright harmful) values for things that have nothing to do with the printer selected, but that are a matter of slicing preferences for the type of model you're printing or hacks that break a lot of things for the sake of making one particular model the user was trying to print come out a little better. For example:

- At least 8 turn coasting on by default. This necessarily produces underextruded, poorly bonding walls.
- Several set max deviation to weird things like a formula based on the layer height or nozzle width.
- Z seam settings (some profiles even have worst-quality setting of "shortest" here)
- Messing up skin preshrink/expand (one has preshrink of 0, expand of 0.8)
- Infill pattern and density
- Whether support interface is on/off (highly model dependent, not at all printer dependent)

I'm not sure how it got like this, but the result is that when giving support to users trying to get their printers working well, folks helping them have to account for all kinds of Cura settings that might have been wrong in one of the profiles that shipped with Cura. And from a nontechnical user perspective, certain printers end up looking bad just because the random flipped switches in the profile Cura shipped make the printer do bad things.

### Expected results

Printer profiles should only be setting things that are actual properties of the printer, and there should be some uniform policy for this: That would include things like the `machine_*` settings, retraction length and speed, default nozzle size, cooling settings based on the fans that are standard on the machine, possibly bridging and overhang settings to go with the available cooling, support overhang angle and support tree angle, anything needed for multi-extruder support, etc.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

`grep -r` on the `resources/definitions` directory for the options I mentioned above (and various others) will quickly reveal the extent of the problem.

Asterchades, 2022-09-19:
Any printer profile for a non-Ultimaker printer is submitted and maintained exclusively by the community. Ultimaker take as much of a hands-off approach as is possible given they lack the facilities to verify what is or isn't well configured (unsurprisingly they only have access to Ultimaker printers). The intent is that the people using these printers will have a better idea of what will and won't work, thus will have a better idea of what does and doesn't need to be set as a default.

To my knowledge the only recent exception to this has been the alteration of the minimum resolution settings with Cura 4.8, where the specified 0.025mm that many printers used was never going to work following the fixes applied to the simplification routines.

Now, with all that said, I personally (as an individual, not a moderator) agree with you that a lot of profiles have default settings that are out of line, ranging from the "mostly harmless" up to the outright problematic. The most egregious I've come across is the Ender-6 profile which sets static values for temperatures rather than letting the material profile set them. These, like any other such values, can be overridden in the print settings - provided the end user actually knows that it needs to be changed differently in the first place.

I'm going to leave this open as a discussion. It doesn't represent a bug, nor does it represent a new feature - though it could qualify as an improvement to an existing feature (that feature being the ability to use and submit custom printer definitions). A community consensus on what should and shouldn't be overridden feels like the best way to go about things, but whatever comes up needs to be completely objective - Cura itself is larger than the team responsible for it, so any changes which could not be reliably automated would represent a significant effort for what is mostly a minimal return.

nallath, 2022-09-19:
I've started cleaning up a few issues I encountered with profiles submitted to us, but it's tedious work. I'm thinking about creating a script that will remove unneeded overrides from the definitions. A lot of machine profiles override default settings with the same default, which bloats them quite a lot (which in turn makes them harder to maintain).

As for the other issues, I think there are a few things that can be done. We should probably consider improving the guides for adding new machines (also; We could use some help there!) and we can have a look at cleaning up existing profiles (We can also use help there!). Cleaning up popular profiles will also help with new profiles, as I've noticed that a lot of people copy existing profiles and thus copy any mistakes that are in there.

richfelker, 2022-09-19:
I think scripts are definitely the right approach here. A good starting point might be a script to enumerate the settings that presently have overrides in one or more printer profiles, along with how many profiles override each. This would quickly identify the most wacky things going on (ones that only appeach once or twice), as well as a candidate set of "reasonable things a printer profile should include" (the ones that appear very frequently). Hopefully the former will be uncontroversial to remove after quick review, and the latter can be the start of a conversation on norms.

richfelker, 2022-09-19:
I suspect the opposite would be more appropriate: an allow-list of things that can go in printer profiles, and a unit test that checks that they don't have anything not on the allow-list.