Ultimaker/Cura #13324: Cant see the print or the printer plate!

Astronate1, 2022-09-18:
### Application Version

5.1.1

### Platform

windows 10

### Printer

Creality Ender 3 

### Reproduction steps

Every time I redownload Cura it works once and then doesn't work again. 
Now I'm trying to get help 

### Actual results

Nothing yet, that's why I'm getting help

### Expected results

I was expecting the print that I was trying to print and the printer plate to show. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I heard somewhere if you delete the roaming file then it would work but I don't know how to find that.

GregValiant, 2022-09-20:
Thanks for the report.
Tell us some more about your computer (Windows Home or Pro, how many monitors you are using, video sub-system, that sort of information.
In Cura under Help is a command "Show Configuration Folder" which will take you to the user information and customization folder.  It is a sub-folder of "Roaming".

MariMakes, 2022-09-22:
Hey @Astronate1 

Welcome to the Ultimaker Cura Github! 🚀 

@GregValiant is right. We need to know more about your configuration before we can help to start troubleshooting.

You can share your configuration folder by:
	1. Go to Help at the top bar of Cura
	2. Select Show Configuration Folder
	3. Select all the folders in there and create a zip OR go up a folder in your folder structure and zip your folder with the correct Cura version
You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser. 

Thanks 👍

closed 2022-10-06 by github-actions[bot]