Ultimaker/Cura #13331: Unable to Slice/Generate Support on Raised Models

madsurfr728, 2022-09-19:
### Application Version

5.1

### Platform

Mac OSX

### Printer

Creality Ender 3 Pro

### Reproduction steps

1. Load .stl file
2. Preferences->Configure Cura->Turn off "Automatically drop models to the build plate"
3. Raise models off the bed to add supports.
4. Use stock profile and generate supports.

After failure.
1. Deleted Cura
2. Deleted Cura folder in "application supports" in the library.
3. Download new file of Cura from website. 
4. No difference.

### Actual results

Slices with 0 time 0g filament and will not allow preview or actually slicing.
[Uploading Cura Issues.zip…]()

### Expected results

Supports should be generated. I was able to do it before with the same profile. Print failed due to adhesion issue so I attempted to replace at a different quality level and that's when it stopped allowing supports to generate above the print bed.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

I recently downloaded a profile from Fat Dragon Games to help with printing some minis and since then, any time I raise the models off the buildplate so I can add supports the program won't slice. Or at least, it slices and it says 0g 0 time. Any ideas on how to fix this or what's going on? It doesn't matter what profile is selected, it does it for all of them. I already deleted the profile from the system and redownloaded the profile fresh but it's still happening. I also deleted cura and the profile from the library and tried using the stock profile without adding any external profiles and I’m still getting the issue. I know I can raise models and slice because I did it prior to this but would love any insight. I also spoke with Joseph Williams from support and he was unable to assist and suggested I reach out

Asterchades, 2022-09-20:
Your project file unfortunately didn't complete uploading before you made your post. I have a feeling that it's going to be necessary as testing a couple of my own models has given the results exactly as you'd expect them, so something is amiss either in the profile you're using or the model itself.

madsurfr728, 2022-09-21:
The problem is that it happens with ANY .stl that I bring in and it doesn't matter what the profile is. All stock profiles do the same thing. I have already completely deleted both the application and all files associated with it as well as downloading and installing a new version.

MariMakes, 2022-10-04:
Hey @madsurfr728,

Welcome to the Ultimaker Cura Github 🚀 
I recognize your issue.

Do you have the experimental setting adaptive layerheights enabled? 
You cannot have an empty first layer with that setting, it will result in 0 minutes and an empty buildplate. 

Can you check for me if it's a duplicate of https://github.com/Ultimaker/Cura/issues/3183
Thank you. 👼