Ultimaker/Cura #13332: Nozzle seize longer Lk4

johnpetrucci17, 2022-09-19:
### Is your feature request related to a problem?

I can't change nozzle seize in the setup of longer lk4 pro (ender 3 pro too)

### Describe the solution you'd like

adding nozzle seize entery

### Describe alternatives you've considered

adding nozzle seize entery

### Affected users and/or printers

longer 3d lk4 pro

GregValiant, 2022-09-20:
The author of the Longer Definition files appears to be "Longer".  Although the definition file states that the nozzles are "variants" there are no nozzle files in the Variants folder.  This would seem to be something you need to contact Longer about.

My printer is an Ender 3 Pro and it has nozzle variants and I can choose from 0.2 to 1.0 nozzle sizes.  So for that printer model this would seem to already exist.

lagout, 2023-10-28:
any update? I have the same problem