Ultimaker/Cura #13336: Slicer crashed

Buddha711, 2022-09-20:
### Application Version

5.1.1

### Platform

Windows 10

### Printer

Tronxy XY-2 Pro

### Reproduction steps

I just pressed slice and it was processing for a about 5-6
[image]
 seconds and crashed with a pop up window.

### Actual results

Slicer crashed pop-up window comes up

### Expected results

Slicer should have successfully sliced the model

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

GregValiant, 2022-09-20:
Thanks for the report.
We will need a Project file and the Cura.log file to investigate this.

Buddha711, 2022-09-20:
I tried to send you the saved project file and your upload section here gave me the error message "We don't support that file type. Try again with a ..."  I saved the project in cura and tried to upload that saved file.  Am I doing something incorrectly there?  How do I get the Cura.log file as well?  Thank you for your help :)

GregValiant, 2022-09-20:
Cura.log can be found by using the "Help | Show Configuration Folder".
Github rules only allow certain file types but one of them is "Zip" files.  Put the Cura.log file and the project file into a zip folder (right click on the file and select "Send To" and then "Compressed (zipped) Folder".

Buddha711, 2022-09-20:
THANK YOU SO MUCH!!!  Here are the requested files
[cura.zip](https://github.com/Ultimaker/Cura/files/9611679/cura.zip)

GregValiant, 2022-09-21:
Much better.
There is a persistent bug in 5.1 that prevents Cura from finishing a slice on Hi-Resolution models.  If you move the model on the build plate and rotate the model then the bug can often be fooled into allowing the slice to complete.
I'll mark this as a duplicate.  I believe this bug has been addressed and will be fixed in 5.2.

After moving and rotating the model it sliced.  (I turned on Tree support to see if it would still slice).
[image]