Ultimaker/Cura #13338: Echec de la découpe avec erreur inattendue

Golay65, 2022-09-20:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Ultimaker S3

### Reproduction steps

Découper

### Actual results

[Doc1.docx](https://github.com/Ultimaker/Cura/files/9605430/Doc1.docx)

### Expected results

?

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

?

GregValiant, 2022-09-20:
Thanks for the report.
Like a slice, it is very thin.

Golay65, 2022-09-20:
[exemple.docx](https://github.com/Ultimaker/Cura/files/9607207/exemple.docx)

GregValiant, 2022-09-20:
With the model loaded in Cura, and with Cura set up and ready to slice, use the "File  | Save Project" command.  Put the 3mf file in a zip folder and post the Zip file here.

Golay65, 2022-09-21:
[ImageToStl.com_ums3_mark4 (1) entonnoir.zip](https://github.com/Ultimaker/Cura/files/9613152/ImageToStl.com_ums3_mark4.1.entonnoir.zip)
Info: sur CURA 4.13.1 j'arrive à découper!!?

GregValiant, 2022-09-21:
Without a Cura "Project File" I can not investigate this.  I don't know what a "*.blend" file is but Cura will not open it.

closed 2022-10-08 by github-actions[bot]