Ultimaker/Cura #13339: Unknown slicing error

Kryszyn, 2022-09-20:
### Application Version

5.1.1

### Platform

Windows 11

### Printer

Anycubic Mega X

### Reproduction steps

I opened a file and clicked Slice

### Actual results

Slicing failed with an unknown error

### Expected results

Cura should generate gcode

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[Cura_Error.zip](https://github.com/Ultimaker/Cura/files/9607850/Cura_Error.zip)

GregValiant, 2022-09-20:
Thanks for the report.
Did you get this view in the Cura Preview stage?
[image]

All those purple areas are errors in the model.
This is the view in Mesh Mixer.  The magenta and blue balls indicate errors.
[image]

I tried using both MS 3D Builder an Mesh Mixer to repair the model but when they finished the model is wrong.  It looks solid with none of those internal structures.
One thing you can try is to go to Special Modes and change "Surface Mode" to "Surface".  It will slice like that but I don't know if it will print as it is supposed to.  There are a lot of errors.
I don't see a bug here.  That's a bad model.  Maybe it was exported from CAD incorrectly?
I'm inclined to remove the bug label but I'll leave it on for a bit and you can respond.

Here is the report from formware.co's STL repair utility:
-> Reading file and indexing vertices
-> Analysed your file:
--> 552 Naked edges (?)
--> 2 Planar holes (?)
--> 3 Non-planar holes (?)
--> 0 Non-manifold edges (?)
--> 0 Inverted faces (?)
--> 0 Degenerate faces (?)
--> 0 Duplicate faces (?)
--> 0 Disjoint shells (?)
-> Repairing: 100.00%
----- Repair completed in 12371ms ------
-> Vertex count changed from 16775 to 13997 (-2778)
-> Triangle count changed from 33264 to 28248 (-5016)

Kryszyn, 2022-09-22:
Thank you for your analysis. I have got gcode files for this project and they work, so I will use is them instead of generating new gcode from STLs. You can close this case.