Ultimaker/Cura #13340: Not slicing, gives a slicing issue

DeDuvel, 2022-09-20:
### Application Version

5.1.

### Platform

Windows 10

### Printer

Creality Ender 5 Plus

### Reproduction steps

1. Load an .STL file in.
2. click slice
3. Error
[image]

### Actual results

[image]

### Expected results

Well... It should slice so I can put it on my card and insert the card in my 3D Printer so it can print the file.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9609237/cura.log)
[CE5P_Tower-of-Insanity-Roof-v1.1.zip](https://github.com/Ultimaker/Cura/files/9609247/CE5P_Tower-of-Insanity-Roof-v1.1.zip)

GregValiant, 2022-09-20:
Than for the report.
This is a duplicate of several other reports where models with no errors in them won't slice.
There is a workaround that sometimes can fool the bug.  Move the model on the build plate and rotate it.
I'll mark this as a duplicate and someone from the Cura team will take a look. Hopefully this will be fixed in 5.2.

I did get it to slice by moving and rotating the model.  The gcode file is 270mb and 8,487,000 lines of gcode.  With your settings it's a 100 hour print.  It also won't display on my anemic laptop.

DeDuvel, 2022-09-21:
Ah okay thanks I will try it tomorrow! Thanks in advance :D

jellespijker, 2024-01-14:
Thank you for reporting this issue, I can confirm that the project file still crashes with our latest internal build 5.7.0-alpha.0+3abb55 . For internal reference the crash is logged crash under Sentry issue: [CURAENGINE-1Z](https://ultimaker-o7.sentry.io/issues/4863584036/)