Ultimaker/Cura #13341: Cura not saving settings

Steven2k6, 2022-09-20:
### Application Version

Arachne_engine_beta_2

### Platform

Windows 11

### Printer

Custom

### Reproduction steps

1. Change Settings in a profile.
2. Click "Update profile from currents settings/overrodes"
3. Go to another profile.
4. Go back to the profile, where you changed some settings.

### Actual results

The Settings aren´t saved.

### Expected results

Settings should be saved.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

-

MariMakes, 2022-10-04:
Hey @Steven2k6,

Welcome to the Ultimaker Cura Github 🚀 

I'm having some trouble reproducing your issue 😞 

Can you try a couple of things for me?
1. Use the latest version of Cura and see if you are still encountering the problem? You can download it here https://github.com/Ultimaker/Cura/releases
2. Check the video in this ticket https://github.com/Ultimaker/Cura/issues/13158 and confirm if this is what you are encountering, or not?
3. Include a video/screen recording of you performing the steps?

Thank you 👼

Steven2k6, 2022-10-05:
This is a downscaled and shortened Video, I uploaded the normal Video on YouTube, before you can't read it.
https://youtu.be/cdTXoMoZgT0

https://user-images.githubusercontent.com/99575824/194061298-f1652a11-10a8-49b3-a7b0-b105694c3eac.mp4

closed 2022-10-05 by MariMakes