Ultimaker/Cura #13343: "Cool Down Speed" Not Timing Correctly

Erin-Stephan, 2022-09-20:
### Application Version

5.0.0

### Platform

Windows 10 on Parallels on macOS Monterey (12.0.1)

### Printer

other - Essentium HSE

### Reproduction steps

Sliced the same simple part at different scales and with different cool down speeds to prove that the final temperature was being inserted at the same corner every time, not changing with the layer time or cooling rate. 

### Actual results

The final temperature inserts at the same location in the print (the front right corner where the infill starts) no matter what part scale or cool down speed I select. I used a wide range or sizes and cool down speeds and nothing changed. 

### Expected results

The final temperature should be getting set closer to the end of the layer as cool down speed increases. Also as scale increases, the final temperature setpoint should not be at the same location. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

This is for an Essentium HSE profile which is not a Cura drop-down printer, it is a custom build with internal profiles so it will not show up in the default Cura 5.0.0 window. I will attach a picture of the various slices and some gcodes. Our gcodes have the setting values listed at the top in the text if you would like to look at the setting values. The picture explains it pretty well.

I am showing the infill layers because it is the most obvious depiction (starting at the start of the infill). The same thing happens on the solid layers, it's just a different location.

[image]

[H08_ToolChangeTest_CoolDownSpeed2_Scale300p.TXT](https://github.com/Ultimaker/Cura/files/9609584/H08_ToolChangeTest_CoolDownSpeed2_Scale300p.TXT)

[H08_ToolChangeTest_CoolDownSpeed35_Scale300p.TXT](https://github.com/Ultimaker/Cura/files/9609591/H08_ToolChangeTest_CoolDownSpeed35_Scale300p.TXT)

[H08_ToolChangeTest_CoolDownSpeed100_Scale300p.TXT](https://github.com/Ultimaker/Cura/files/9609597/H08_ToolChangeTest_CoolDownSpeed100_Scale300p.TXT)H08

[H08_ToolChangeTest_CoolDownSpeed100_Scale500p.TXT](https://github.com/Ultimaker/Cura/files/9609613/H08_ToolChangeTest_CoolDownSpeed100_Scale500p.TXT)

[H08_ToolChangeTest_CoolDownSpeed35_Scale100p.TXT](https://github.com/Ultimaker/Cura/files/9609639/H08_ToolChangeTest_CoolDownSpeed35_Scale100p.TXT)

ZacharyHowell-Essentium, 2022-09-26:
@nallath, is this issue currently being tracked?
Thanks!

Erin-Stephan, 2022-09-29:
I have updated the ticket to include a PDF presentation of the issue and also the project file with Ultimaker settings to show that it is also occurring on Ultimaker slices. I am available for a call if you need more clarity on the issue I'm getting. Thanks

MariMakes, 2022-10-05:
Hey @Erin-Stephan,

Welcome to the Ultimaker Cura Github 🚀 .
This research is just amazing 😮 
Thanks for all the hard work you did on this. 🎉 

[image]

Fun fact, the problem that the M104 command is inserted at the incorrect position in the gcode has already been reported internally and we've already added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-9540

These test cases will make resolving probably even easier. 
Thanks for the report! 👍

Erin-Stephan, 2023-01-17:
Hello!

Can I get an update on this issue? Has it been resolved in the new releases of Cura? I read all the bugfixes for the most recent releases and I don’t see any mention of it and I’m about to do a whole bunch of testing on tool changes, where the heat up and cool down settings are going to be very useful.

Thanks,
Erin

Erin Stephan
Process Engineer | Essentium, Inc.
***@***.******@***.***>
C: 713.859.3686
Essentium.com<http://www.essentium.com/>

[Black text on a white background  Description automatically generated]

[Image removed by sender. giphy (13)]<https://user-images.githubusercontent.com/40423138/194056624-cb962987-0890-4e21-aae2-4f02d858bff6.gif>

EXTERNAL EMAIL: Be cautious, do not click links/attachments. Never give out your user ID or password. Verify the recipient's validity

Company Disclaimer: The information contained in this communication is confidential, private, proprietary, or otherwise privileged and is intended only for the use of the addressee. Unauthorized use, disclosure, distribution, or copying is strictly prohibited and may be unlawful. If you have received this communication in error, please notify the sender immediately.

MariMakes, 2023-01-17:
Hey @Erin-Stephan,

Some significant work has been done on this! I believe we just forgot to update this issue. 🙈 

We expect this Pull Request to be shipped with Cura as soon as it passes the review🎉 
https://github.com/Ultimaker/CuraEngine/pull/1779

Thanks again for reporting this 💪

Erin-Stephan, 2023-01-31:
Hey @MariMakes thanks for the quick reply. Do you have any update or insight on when the next shipment of cura settings will go out?