Ultimaker/Cura #13345: 5.1.1 is not launching on Catalina

stevemaser, 2022-09-20:
### Application Version

5.1.1

### Platform

macOS 10.15.7

### Printer

doesn't matter

### Reproduction steps

Drag copied Cura 5.1.1 to a managed 10.15.7 system.  

### Actual results

 It did not launch when double-clicked, so I tried opening it through Terminal and got this:

maser@diamond ~ % /Applications/Ultimaker-Cura.app/Contents/MacOS/Ultimaker-Cura 
[1979] Error loading Python lib '/Applications/Ultimaker-Cura.app/Contents/MacOS/libpython3.10.dylib': dlopen: dlopen(/Applications/Ultimaker-Cura.app/Contents/MacOS/libpython3.10.dylib, 10): Symbol not found: _preadv
  Referenced from: /Applications/Ultimaker-Cura.app/Contents/MacOS/libpython3.10.dylib (which was built for Mac OS X 11.6)
  Expected in: /usr/lib/libSystem.B.dylib
 in /Applications/Ultimaker-Cura.app/Contents/MacOS/libpython3.10.dylib

The app launches as expected on macOS 11 and up.  

### Expected results

That it should launch normally on 10.15.

That said, 10.15 is probably unsupported by Apple now, so it's not critical, but we wanted to pass this on.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

nothing

chaimleib, 2022-09-25:
I'm getting the same issue, also on macOS 10.15.7, using Cura 5.1.1. The icon bounces once, and then stops. It doesn't matter whether I install via the DMG or via homebrew.

Cura 5.1.0 launches fine.

MariMakes, 2022-10-04:
Hey @chaimleib,

Unfortunately, We had to make a conscious decision to stop supporting MacOS 10.15 since our build tools also stopped supporting that MacOS version, just like Apple. 

A list of our supported Operating Systems can be found here: https://support.ultimaker.com/hc/en-us/articles/360011889579-On-what-operating-systems-is-Ultimaker-Cura-available-

I'll be closing this issue since it's intended behavior. 
Thanks again for the report 👍