Ultimaker/Cura #13348: Added Kobra Max to 5.0. slicing failed

ZondervanAxle, 2022-09-21:
### Application Version

5.0.0 and 5.1

### Platform

windows 10

### Printer

Anycubic Kobra Max

### Reproduction steps

kobra max specs are 400x400x450
started a stl for the helmet of Sauron from lord of the rings
pressed slice and it failed. happens both in 5.0 cura and 5.1

### Actual results

failed slice

### Expected results

slice
[CFFFP_Helmet.zip](https://github.com/Ultimaker/Cura/files/9612572/CFFFP_Helmet.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

let me know if its just me, im new and this is a new printer. i had no issues with my Mega x

GregValiant, 2022-09-21:
Thanks for the report.
No, it isn't you.  There is a bug that keeps some models from slicing even if the models are like this one and have no errors.
There is a workaround that can help.  Move the model on the build plate and try rotating it.  I did get it to slice.
I'll mark this as a duplicate.  Someone from the Cura team will take a look.
BTW with your settings this came out to be a 434 hour print.  That's 3 weeks.
Here is the orientation that sliced.  (My old laptop won't show a preview.)
[image]

ZondervanAxle, 2022-09-22:
that is awesome! thankyou. im still new to 3d printing and yeah my settings
arent great yet. that would be kinda funny in a bad way to print for 3
weeks.

jellespijker, 2024-01-14:
Thank you for reporting this issue, I can confirm that the project file still crashes with our latest internal build 5.7.0-alpha.0+3abb55 . For internal reference the crash is logged crash under Sentry issue: [CURAENGINE-1X](https://ultimaker-o7.sentry.io/issues/4863502807/)