Ultimaker/Cura #13352: Slicing Error

skbnjit, 2022-09-22:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Ultimaker 3

### Reproduction steps

I clicked 'slice' on Cura to slice my print and it went about 80% through before popping up with an error saying slicing error

### Actual results

N/A

### Expected results

It should've sliced

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[QM2.zip](https://github.com/Ultimaker/Cura/files/9621189/QM2.zip)

[image]

GregValiant, 2022-09-22:
Thanks for the report.
Here is a report on the model:
--> 0 Naked edges (?)
--> 0 Planar holes (?)
--> 0 Non-planar holes (?)
**--> 18 Non-manifold edges (?)
--> 21 Inverted faces (?)
--> 502 Degenerate faces (?)
--> 8 Duplicate faces (?)**
--> 0 Disjoint shells (?)
That's why it won't slice.  When a model is not "watertight" Cura can't figure out what is the inside and what is the outside.
The view below is from Mesh Mixer.  You can see how many errors there are.  Mesh Mixer's attempted repair resulted in parts of the model disappearing.
I'm going to remove the bug label as it appears to be a model problem.  I'll leave this open in case you want to respond or if you think that is premature.
[image]

schlundl, 2022-09-24:
That’s a silly response to “fix” the issue. This has been happening to me as well and no matter what file I upload on widows 10 the slicing error appears. For the little boat from thingiverse, to my own blender models. Sooo many people have been complaining of slicing error, and you guys have been useless

GregValiant, 2022-09-24:
@schlundl - What makes it a silly response?  The model is a mess, is non-repairable, and if it does slice the slice is wrong.  The only fix for this issue is to repair the model.

Here it is in PrusaSlicer.  The stern of the model looks nothing like that actual model.
[image]

schlundl, 2022-09-24:
Well if you look at the complaint, it is a bug. On cura. Not pruca. And the problem isn’t with the model, it’s cura unfortunately. Bug going around and it needs fixed. Sure theirs things that can be fixed with the model but that’s not the problem

closed 2022-10-08 by github-actions[bot]