Ultimaker/Cura #13353: Incomplete raft

Tettink, 2022-09-22:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Ultimaker s5

### Reproduction steps

1.Loading in files, merge and position them x0 y0 z0
2. Generate with default settings + raft ultimaker ABS material black and grey.

### Actual results

Ultimaker skips half of the 3e layer of the raft
[image]

### Expected results

[image]
The only way for me to fix this with this particular file is turning it a bit.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Files.zip](https://github.com/Ultimaker/Cura/files/9623130/Files.zip)

GregValiant, 2023-01-07:
There is [this thread](https://community.ultimaker.com/topic/30374-2nd-layer-above-raft-is-missing/#comment-317178) on the UM Forum that also describes this behavior.  Once again, moving the model on the Cura build plate caused the problem to go away.
Here is the project file from that thread.
[Partial Raft Settings.zip](https://github.com/Ultimaker/Cura/files/10364615/Partial.Raft.Settings.zip)

Tettink, 2023-01-07:
Hi,

Yes it indeed it did, sorry for not responding earlier.

Verstuurd: zaterdag 7 januari 2023 01:08
Aan: Ultimaker/Cura ***@***.***>
CC: Tettink ***@***.***>; Author ***@***.***>

MariMakes, 2023-01-09:
Hey @GregValiant and @Tettink,

Thanks for your report 👍

@GregValiant, I double-checked but could not resolve the issue by moving it away from the center of the buildplate. Do you have a screenshot or video of the project file where we can see that the top layer of the raft becomes complete again when you move the model? 😇 

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/12276
We have a ticket on our backlog with the intent to improve this behavior.
I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/12276

closed 2023-01-09 by MariMakes