Ultimaker/Cura #13358: Bug report

JBCrafting, 2022-09-23:
### Application Version

5.1.1

### Platform

windows 11

### Printer

Ender 5 plus

### Reproduction steps

I placed the item on the build plate, arranged, set infill to 10%, sliced....... slice failed error 

### Actual results

slice failed     

### Expected results

expected it to slice the file for print

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

did the update and nothing works program closes during slice, fails to slice, fails to load.... have not been able to print a file yet. files paid for and said to be tested sanix3d.com

Asterchades, 2022-09-23:
Ticking the checkboxes doesn't attach the corresponding files. You need to do that yourself, as described in the form used to submit a bug report in the first place. Note that it also needs to be done through the GitHub web interface - it cannot be done using the e-mail system.

Without at least the project file no diagnostics can be done. I would recommend finding a non-paid file which experiences the issue (you claim "nothing works" so that shouldn't be hard) to upload in the interest of not potentially violating any form of agreement between you and the seller.

JBCrafting, 2022-09-23:
Its ok I used the Creality slicer software and got the file to print
Nothing will slice with the cura software after the 5.1.1 update I have never had an issue till this update. Maybe someone that knows how to complain better than I do can get yall what you need to help you with fixing the issue.  Sorry I cant be of anymore help to you.

***@***.***

schlundl, 2022-09-24:
You all need to fix the bug issue! Slicing fail

JBCrafting, 2022-09-25:
> You all need to fix the bug issue! Slicing fail

It is easier for them to treat us like we are stupid and be smart tails than to figure out why something is not working. his/her comment made it seem as if I had just started printing today and he was printing before he/she was born.

MariMakes, 2022-10-05:
We are constantly working on making Ultimaker Cura better. 
Our community is a big part of that by requesting new features and reporting bugs. 
Let's be kind to each other. We can only help if we work together. 

[image]

MariMakes, 2022-10-05:
But @Asterchades is right. We need a project file from you, @JBCrafting, to help you troubleshoot. 

Do you have a project file for us?
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser. 

Thank you 👼

closed 2022-10-19 by github-actions[bot]