Ultimaker/Cura #13359: Cura Causes Entire Computer to Crash

shtarbanov, 2022-09-23:
### Application Version

5.1.0

### Platform

Windows 11

### Printer

n/a

### Reproduction steps

Suppose you have a folder with hundreds of STL files and you want to open one of them in Cura. However, you accidentally select them all and press Enter to open the file in cure. But now, each selected file opens its own instance of Cura, resulting in  HUNDREDS of Cura windows opening one after the other, causing the computer crash. 

### Actual results

Entire computer crashes

### Expected results

When multiple STL files are selected, they should all be opened in a single instance of cura. The default expectation of the user is that if they have multiple STL files selected, the user is probably trying to put the selected files on the same build plate. The behavior should thus be the same as if user is importing the files into cura. Only one instance of cura should be opened, regardless of how many files are selected.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

n/a

MariMakes, 2022-10-05:
Hey @shtarbanov,

Welcome to the Ultimaker Cura Github 🚀 
Sorry, it took us a while to get back to you. 😞 

Ugh, I'm sorry to hear that your computer crashed because you were just trying to open a single STL.
We have a setting in our preferences that prevents multiple instances of Cura from opening.
It's called **Use a single instance of Cura**. 
[image]

Can you let us know if this is the solution you are looking for?
Thank you 😄 

If this is not what you are looking for I'm not sure how if this is something we can fix with Cura. 😞

shtarbanov, 2022-10-05:
It is great that you have this setting. It's not so great that it's
something nonobvious and something one would not even consider looking for.

So how about the following solution. When someone tries to open multiple
selected stl files a window pops up asking the user whether they want to
open the files in multiple instances or in a single instance. And then in
that window they also have the option to have their choice remembered for
next time.

How does that sound?

MariMakes, 2022-10-05:
Hey @shtarbanov 

Good to hear back from you. 

As mentioned before, what happens when you open a file from your windows file explorer is not something that we can change in Cura if the preferences are not set correctly. I brought it up to the team and we believe we will not work on improving this anytime soon and will therefore close this issue.

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.
I hope you understand 😇

closed 2022-10-05 by MariMakes