Ultimaker/Cura #13362: Slicing error

3dprinting-source, 2022-09-23:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

ender 3 v2

### Reproduction steps

Tried to slice

### Actual results

Failed slice error

### Expected results

Slice file

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9633312/cura.log)
[CE3PRO_key_chain_X15.zip](https://github.com/Ultimaker/Cura/files/9633317/CE3PRO_key_chain_X15.zip)
[image]

GregValiant, 2022-09-23:
Thanks for the report.
This sliced on the first try for me and I can't duplicate the problem.  Is this happening to you with any model or just this specific grouping?
You can try rotating the group 90° on the build plate.  There is a slicing bug that rotation and movement seems to nullify.

schlundl, 2022-09-24:
This happens to me too. No matter what file I load, please help!! I have even tried reinstalling cura

schlundl, 2022-09-24:
Also tried rotating and all that

GregValiant, 2022-09-24:
@schlundl could you open another ticket please?.  The Cura team will want to see your log files and it sounds like you have a different problem than @3dprinting-source is having.

closed 2022-10-08 by github-actions[bot]