Ultimaker/Cura #13363: Cura 5

ledinger, 2022-09-23:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

CR 10

### Reproduction steps

Slice
[S 02a.zip](https://github.com/Ultimaker/Cura/files/9633487/S.02a.zip)

### Actual results

Error Message.
[S 02a.zip](https://github.com/Ultimaker/Cura/files/9633485/S.02a.zip)

### Expected results

Objekt File.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9633513/cura.log)
[CFFFP_S 03a.zip](https://github.com/Ultimaker/Cura/files/9633527/CFFFP_S.03a.zip)

GregValiant, 2022-09-23:
Thanks for the report.
Your S 02a model sliced with no problems.  In the project file I had to rotate the model 90° to get it to slice.
There have been numerous bug reports regarding perfectly fine models that don't want to slice when at the midpoint of the Cura build plate.  Moving and/or rotating the model often fools the bug.
I'll mark this as a duplicate.

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release

[image]

closed 2024-01-14 by jellespijker