Ultimaker/Cura #13365: slicing failed for interface settings

oddwatcher, 2022-09-23:
### Application Version

5.1.0

### Platform

Windows 11

### Printer

Ender3

### Reproduction steps

[image]
tried to use this support setting keep rest default while print a screw horizontally (means a lot overhung)
[image]

### Actual results

the slicer refuse to work stuck at 2/3 and stopped  
[image]

### Expected results

Giving the support all the way up to the model with out the interface

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[CE3_NanoViseV2_CLAMP_SCREW_DeskClamp_Version.zip](https://github.com/Ultimaker/Cura/files/9634525/CE3_NanoViseV2_CLAMP_SCREW_DeskClamp_Version.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/9634534/cura.log)

GregValiant, 2022-09-23:
Thanks for the report.
I couldn't get your model to slice either and it is watertight and shows no errors.  After moving it and playing with it I finally rotated it 45° and like magic, it sliced.  This is a known bug so I'll mark this as a duplicate.

Enough of that.  I've printed a fair number of screws and my preference is to print them vertically.  I slice the model twice, once for the base at .2 layer height and the other for the threads at .1 layer height.  Then I combine the files.  I've found that the threads come out much better and there will be no scars from support.
Although there were no errors found in this model I uploaded it to [THIS SITE](https://formware.co/OnlineStlRepair) to see what it could do.  It didn't find anything wrong either, but it did add faces and vertices.  The altered model slices in any orientation.
This is the original model rotated 45°.
[image]

I'm attaching the altered model as well as a gcode I put together.  (I've written a script to combine gcode files so there wasn't any actual work involved).  The slices for the gcode file were done with my Ender 3 Pro active so the StartUp Gcode will be mine rather than yours.  I don't have an ABL so the G92 command is not in there.  If nothing else the preview of the gcode in Cura is interesting when viewed from the front in orthographic view as you can see the layer height change.
[GV Screw_Files.zip](https://github.com/Ultimaker/Cura/files/9636354/GV.Screw_Files.zip)

oddwatcher, 2022-09-24:
Thanks a lot. Why i print these screws in horizontal is for the strength. My PLA filaments somehow don't stick firmly between layers . the result from the default settings are fine out of the printer ,but if you apply a slightly higher force on the side it will break since screws mainly provide its strength on the vertical side i have to put it down.
Anyway thanks for your help it is amazing to have a reply so quickly.

closed 2022-09-24 by oddwatcher