Ultimaker/Cura #13366: Cura does not run on Void Linux

sinetek, 2022-09-23:
### Application Version

5.1.1

### Platform

Void Linux x86_64

### Printer

.

### Reproduction steps

I got the release `Ultimaker-Cura-5.1.1-linux.AppImage`. Tried to run it and get this error.
It happens when I open a file.

```
PyInstaller/loader/pyimod02_importers.py:499: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
Cyclic dependency detected between "file:///home/phil/Downloads/ultimaker/squashfs-root/share/cura/resources/qml/Actions.qml" and "file:///home/phil/Downloads/ultimaker/squashfs-root/share/cura/resources/qml/Actions.qml"
QQmlEngine::setContextForObject(): Object already has a QQmlContext

(Ultimaker-Cura:3551): Gtk-WARNING **: 21:49:57.667: Could not load a pixbuf from icon theme.
This may indicate that pixbuf loaders or the mime database could not be found.
**
Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/Adwaita/16x16/status/image-missing.png: Unrecognized image file format (gdk-pixbuf-error-quark, 3)
Bail out! Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/Adwaita/16x16/status/image-missing.png: Unrecognized image file format (gdk-pixbuf-error-quark, 3)
fish: Job 1, './Ultimaker-Cura' terminated by signal SIGABRT (Abort)
```

I tried to build Cura manually but it's really complicated and I don't understand how. I don't want to use the conan thing because it hardcodes dependenciens on dpkg/apt and I don't have that.

### Actual results

The file browser should work.

### Expected results

The file browser should work.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9636769/cura.log)

MariMakes, 2022-10-05:
Hey @sinetek,

Thanks for your report. 👍 

I'm not running Linux myself, but have you try the Linux Modern builds? 
I've heard that they solve a number of problems for Linux users who are on modern distro's. 

Just look for an app image that ends in -linux-modern.Appimage
https://github.com/Ultimaker/Cura/releases

Can you let us know it this works for you? If not, I can bring it up to the team 💪

sinetek, 2022-10-05:
Unfortunately the issue persists with a newer glibc, and it seems to be gtk
problem.

thanks for your time, i appreciate it. Fortunately we have cura 4 so I will
continue using that.

MariMakes, 2022-10-13:
Hey @sinetek 

Sorry, it took us a while to get back to you 😞 

I'm not running on Linux, I'm on Windows.
But I asked the team and they suggested taking a look at this GitHub thread:
There are some good tips in there for Linux users. 
https://github.com/Ultimaker/Cura/issues/12266

Can you let us know if you are still stuck? Thank you 🙏

closed 2022-10-28 by github-actions[bot]