Ultimaker/Cura #13370: Slicing issue

OhPetyaexe, 2022-09-24:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Ender 5 Pro

### Reproduction steps

Tried to slice one stl. from Thingiverse.

### Actual results

 The slicing fails and drops a Bug Report button.

### Expected results

Should've sliced.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9638268/cura.log)

schlundl, 2022-09-24:
Mine does this too. No matter what file I load. It will keep stopping low to mid slice reporting unexpected error, please report bug.

MariMakes, 2022-10-05:
Hey @OhPetyaexe,

Welcome to the Ultimaker Cura Github 🚀 
Sorry it took us a while to get back to you 😞 
Thanks for your report and your cura log, but we need at least your project file to help with troubleshooting. 

Do you have a project file for us?
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser. 

Thank you 😄

closed 2022-10-19 by github-actions[bot]