Ultimaker/Cura #13371: (5.1.1) Incorrectly generates g-code.

ArdjunaMara, 2022-09-24:
### Application Version

5.1.1

### Platform

Windows 10

### Printer

Custom FFF printer

### Reproduction steps

Layer 170 incorrectly prints one wall line. It can be seen on the print animation (last wall).

### Actual results

[image]

### Expected results

I would like the line to be printed continuously.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[B_and_R_cleaner_fitting.zip](https://github.com/Ultimaker/Cura/files/9638668/B_and_R_cleaner_fitting.zip)

GregValiant, 2022-09-24:
Thanks for the report.
When I slice your 3mf file (thanks for that) that layer is full so I'm not able to reproduce the problem.
[image]

Could you post a gcode file of that slice?  I can read it into AutoCad (to take Cura out of the loop) and see if the lines are there.  If they aren't then it does look like a slicing bug.  If they are in the gcode but not in the preview then it is likely a display issue (maybe a bug there).

ArdjunaMara, 2022-09-25:
There is no error. I understood the reason. That line is not a wall, but a
roof. And with the monotonic top mode turned on, the nozzle jumped. Thanks
for the prompt response.
 B_and_R_cleaner_fitting.gcode
<https://drive.google.com/file/d/1mxYP4hIs7zJDoq62pKoA3oy9aHcT23lg/view?usp=drive_web>
[image: image.png]

GregValiant, 2022-09-25:
Since it's resolved can we close this ticket?

ArdjunaMara, 2022-09-25:
Of course.

closed 2022-09-26 by GregValiant