Ultimaker/Cura #13372: Project name reverts after saving gcode

slab94, 2022-09-24:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Creality Ender 5

### Reproduction steps

1. Start Cura
2.  drag and drop stl file into Cura
3. Slice stl file and save gcode file.
4. Saved project file (3mf & html)
5. deleted current sliced object (either ctrl-D or Delete key)
6. drag and drop new stl file into Cura
7. Slice and save gcode file
8. Project name reverts to prior name 

### Actual results

See #8 above. Note. Problem occurs when either type of Project file is saved (3mf or html or 3mf & html) See attached images.
[image]
[image]
[image]

### Expected results

I expect the name (CE_wheel_pivot_V2) shown in Step #6 to remain until a new gcode file is loaded.

If this is considered normal, than I would request a new feature to allow user configuration of this.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I believe that the above documents the problem. I consider this very aggravating, but does not affect the output quality, other than unintentionally overwriting a project file.

GregValiant, 2022-09-24:
Thanks for the report.
This is a duplicate report and I know I can reproduce it.  I see you are using 5.0 but this still happens in 5.1.
You can look at #11756, #11803, #13095, and on the backlog as [CURA-8897](https://ultimaker.atlassian.net/browse/CURA-8897)
I also view this as a productivity issue.  Having to duplicate work because of an over-write by Cura is not good.

slab94, 2022-10-04:
You may have already seen this, but if you save the project before saving the gcode, then the project name stays (until you load a new stl and save the gcode first). Hopefully this helps.

rogerw-gh, 2023-01-11:
This has been happening since at least 4.12.1 on Linux, and is still happening on the latest build [5.2.1] , similar workflow:
Open a 3MF file, process,  open a new 3MF file, pull some STL's in, remove some STLs, slice, filename reverts.
What is deeply frustrating is that the filename is cleared when saving the GCODE, I avoided Cura 5 for that reason and I shall avoid 5.2.1 for the same reason.  I expect the filename to persist into the Save to Disk dialogue. Platform - Ubuntu 20.04.05 LTS.