Ultimaker/Cura #13374: Slicing Failed Unexpected Error

hhagquist, 2022-09-24:
### Application Version

5.1.0

### Platform

Windows 11

### Printer

Ender 3 Pro

### Reproduction steps

Tried to slice this stl and got the error

### Actual results

Received slicing error

### Expected results

Should create the gcode

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Included Project File and Screenshot
[Vader_Dlx_Planter_50mm.zip](https://github.com/Ultimaker/Cura/files/9639270/Vader_Dlx_Planter_50mm.zip)

schlundl, 2022-09-24:
same happened to me. still no fix

GregValiant, 2022-09-24:
Thanks for the report.
This sliced for me on the first try so I can't duplicate the problem.
[image]

There have been several bug reports (that are confirmed slicing issues) involving models like this one that are complex and error free.
A workaround that can sometimes fool the bug is to move the model on the build plate and/or rotate the model.
I'll mark this as a duplicate and as a slicing error.  I believe this will be fixed when 5.2 is released.

hhagquist, 2022-09-24:
Rotating it 90 deg solved the slicing issue.

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release

[image]